betaflight/betaflight #8079: On Matek f722 se oled Feature working not! out off the box

brainbubblersbest, 2019-04-23:
In cli Parameter set dashboard_i2c_bus = 1
Need configured manually. 
I write this here because i found nothing documented about this Parameter for the oled Display feature. Searched for an entire day to get this to work. Maybe matek f722 se build need a slight Modifikation.

etracer65, 2019-04-23:
No, it's not appropriate that this is a default. First, very few people use an OLED module. Second, the I2C connections can be used for other (much more likely) things like a magnetometer or external barometer.

closed 2019-04-23 by etracer65

jflyper, 2019-04-23:
Tbh, OLED display has its own default constant `DASHBOARD_I2C_INSTANCE` while default bus for mag and baro have their own constant `BARO_I2C_INSTANCE` and ` BARO_I2C_INSTANCE`. So, in theory, `OLED display has its own constant `DASHBOARD_I2C_INSTANCE` could be defined for MATEKF722SE target.

However, I don't bother to do it as setting it via CLI will be a standard once we completely migrate to unified target.

brainbubblersbest, 2019-04-23:
I Simply would share my experience. I have magnetometer on this connection
too. And on some F4 fcs baro mag and oled.
That Display is a nice Feature for a few bucks. Maybe a few lines in
description on betaflight docs, about dashboard parameters would be enough.

It's just a hint. BTW Thanks for your Work on this Firmware.