betaflight/betaflight #8080: CL Racing F7 giving wrong current values

frankpv, 2019-04-23:
I have 2 identical quads. Both running CL Racing F7 flight controllers and Tmotor F55A Pro ESC. One is on 3.5.7 and has a correctly working current draw reading in the OSD. The other is on 4.0.0 and reads 42 amps as soon as I arm it. That's 42 amps just sitting there idling.  During flight the current doesn't move much either. 

I will post a cli dump later tonight. 

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

 - CL Racing F7, Tmotor F55A Pro, Xing 2306 1800kv, Rush Tank VTX, TBS Unify HV, R9mm, Runcam Micro Eagle, 6S 1050mah.

frankpv, 2019-04-24:
Updated with diff

etracer65, 2019-04-24:
You haven't set the amperage meter scale (`ibata_scale`) and it's still at the default of 250 for that flight controller. You need to set it appropriately for the current sensor on that ESC based on information provided by the ESC manufacturer.

It's also possible you have miswired the current source. I would check the voltage on the current source pin coming from the ESC to see if it's around zero volts when just powered by the battery and not armed (REMOVE PROPS for safety!). It won't be zero volts but should be a very low value.

bnn1044, 2019-04-26:
@frankpv this current sensor scale depends on what ESC do you use. if you setup the wiring correctly. then you have to either ask the ESC manufacturer about the current scale number or you have to find it out by calibrating. it have nothing to do with the hardware in the FC. it just pass through voltage to the F7 chip.

frankpv, 2019-05-04:
We can close this issue. It appears to be a hardware issue. I replaced the FC with a new one and current looks good!

closed 2019-05-04 by frankpv