betaflight/betaflight #8083: CLRACINGF7 does not boot with 4.1.0 #1577 and bidircfg/.../DEFAULT.cf settings

danielschafer, 2019-04-23:
**Describe the bug**
When BF 4.1.0 #1577 is flashed to the CLRACINGF7 target and the bidircfg/.../DEFAULT.cf are applied. The FC does not boot anymore!

**To Reproduce**
Flash BF 4.1.0 #1577 to CLRACINGF& target and apply these settings: bidircfg/.../DEFAULT.cf

**Expected behavior**
RPM filter is working.

**Flight controller configuration**
Everything default, like described.

Docteh, 2019-04-23:
1577 appears to be a particular Jenkins build of betaflight, not PR #1577 from Nov 2016
https://ci.betaflight.tech/job/Betaflight/1577/

https://github.com/betaflight/bidircfg/blob/master/DEFAULT.cf

Since its a jenkins build and not a release, maybe try and track down if a particular line is causing it to not boot?

danielschafer, 2019-04-24:
It is this line:

set dshot_bidir=on

Every other lines work.