betaflight/betaflight #8086: Airbot F4 DSMX not recognized.

jamesgfeller, 2019-04-23:
**Describe the bug**
Airbot F4 DSMX not recognized. Nothing moves in receiver tab.  Reflash to 3.5.7 and all works normally again.  Reflashed 4.0 and same problem.

# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.26V, Core temp=35degC
Stack size: 2048, Stack address: 0x1000fff0
Config size: 3021, Max available config: 16384
GYRO=MPU6000
System Uptime: 24 seconds, Current Time: 2019-04-23T22:30:56.023+00:00
CPU:6%, cycle time: 127, GYRO rate: 7874, RX rate: 33, System rate: 9
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
Arming disable flags: RXLOSS CLI MSP

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

```# diff

# version
# Betaflight / AIRBOTF4 (AIR4) 4.0.0 Apr 11 2019 / 06:55:53 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

board_name AIRBOTF4
manufacturer_id 

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature TELEMETRY

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 900 1400 0 0
aux 1 13 0 900 1400 0 0
aux 2 35 2 900 1400 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 3 h

# display_name

# master
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set serialrx_provider = SPEK2048
set motor_pwm_protocol = DSHOT600
set align_board_yaw = -90

# profile
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 150
set pitch_rc_rate = 150
set yaw_rc_rate = 150

# end the command batch
batch end```

jflyper, 2019-04-24:
Tried to reproduce with Airbot F4 (ICM20601), freshly flashed with 4.0.0 target AIRBOTF4, loaded with your config diff. A satellite receiver was connected using the dedicated connector.

It worked straight from the first reboot.

I think you have a configuration problem. Please retry from full-chip erase.

watageek, 2019-04-24:
It might be linked with the issue  #7852.

Adding The conditionnal in driver.h solved the issue. ( most likely a clone FC). #7852

watageek, 2019-04-24:
sorry it was in timers.c
I did jumped to conclusions swiftly, please try this HEX to see if it solve your issue.

[betaflight_4.0.0_AIRBOTF4.hex.zip](https://github.com/betaflight/betaflight/files/3112476/betaflight_4.0.0_AIRBOTF4.hex.zip)

jflyper, 2019-04-25:
@jamesgfeller What is the nature (manufacturer and model) of your satellite receiver?

jamesgfeller, 2019-04-25:
That hex worked perfectly.  Rx works as expected and arms.  Have not flow but will fly tomorrow and report back. Thank you so much for that.  I wont even pretend to understand what the timer.c is but thanks so much.  Is this something that can be integrated going forward on Betaflight releases?
Does this effect only uart one?

Satellite is a lemon Rx DSMX and have never had any issues.  Board is a Ready made RC airbot F4.

jflyper, 2019-04-25:
@jamesgfeller This has nothing to do with timers; rather, it is related to the initial pin state when the board is booted.

Apparently, some compatible satellite receivers don't like the new initial state.

@watageek Reverting the initialization is not the solution.

We need to investigate what is going on...

watageek, 2019-04-25:
Sorry @jflyper
I am not really good with coding...
But I can do a copy/paste and compile, and you know where to look. 

Providing a temporary fix is the only thing I can do.

mikeller, 2019-05-12:
@jamesgfeller: Can you please check if the following solves the issue for you:

```
resource pinio 1 b8
set pinio_config = 1,1,1,1
```

(from https://github.com/betaflight/betaflight/issues/7849#issuecomment-491553115)

jamesgfeller, 2019-05-12:
@mikeller:  tried new 4.02 version and made changes above.  Same issue.  No rx movement.

mikeller, 2019-05-12:
@jamesgfeller: Thanks for testing - now we at least know it's not the same issue as in #7849.
@jflyper: Can you get your hands on a schematic for the AIRBOTF4?

closed 2019-06-19 by stale[bot]