betaflight/betaflight #8092: Failsafe don't work on RX disconnected

emiliorg81, 2019-04-24:
Hi.
I loose a racer drone because I think the ibus receiver was disconnected during I was flying...
Failsafe don't apply if RX is disconnected I suppose. I use Ibus.
Is there any configuration to apply failsafe in this case? Is it possible apply failsafe if your altitude is more that an specific meters?
Thanks.

etracer65, 2019-04-24:
If the receiver was physically disconnected (wires broken) in flight, then yes it should cause a failsafe. Whether or not a failsafe is triggered in Betaflight when you lose radio signal between the TX/RX is determined by your transmitter and receiver settings. You should always test your failsafe settings before actual flight to ensure they are working and behave as expected.

emiliorg81, 2019-04-24:
Thanks.
I try with sbus and it seems work better than ibus.

closed 2019-04-24 by emiliorg81