betaflight/betaflight #8094: timerInit IOConfigGPIOAF strikes back again from a backdoor!!!

jflyper, 2019-04-25:
As reported several times in different context, some compatible satellite receivers has a problem with the initial state of the pin.

#7852 
#8086 
Some post in FB

It has been tracked down to #6681.

I suspect that these receivers are sensitive to some state or state transitions of the pin.

watageek, 2019-04-25:
Add#8021

jflyper, 2019-04-25:
@watageek Do you aware of a case on non-Omnibus F4 and non-Airbot F4 case?

I would also like to know how the receivers are connected.

It may have something to do with SBUS/PPM shared architecture of these boards.

watageek, 2019-04-25:
At the moment, no. @jflyper

For me, the redcon satellite was hook on the dsm port ( uart1) connector on a Omnibusf4.

jflyper, 2019-04-25:
@watageek Is that the original Omnibus F4, with SBUS and PPM 0-ohm jumpers?

watageek, 2019-04-25:
@jflyper 
No it was a clone at 12usd.

It has 3 pads and I bridged mine like on the picture.
[image]
[image]

[image]

The photo come from their website.

jflyper, 2019-04-25:
Do you have any instruction / idea / description of what these resistors are?

[image]

watageek, 2019-04-25:
@jflyper

No sorry. No instruction. Nothing except the board. No connector either. 

I have 0ohms accross both with a classic DMM tested on the board. But the board is messed up. I can desolder them if you want. But if they are low Ohms I won't be able to measure them.

jflyper, 2019-04-25:
Nah, the actual resistor, if possible, that could be removed is between USB and DSM connector lol

jamesgfeller, 2019-04-27:
Here is info on my board.  Lemon rx non diversity connected  on uart 1 with connector shown. Hopefully this helps.

[image]

jflyper, 2019-04-27:
We need someone with a logic analyzer/oscope and setup that has the same issue to see what's going on at the boot...

Specific items that should be investigated are:

(1) Find out the state of the receiver; is the receiver emitting any signal when it is bound to a transmitter?

(2) If it is, then is it a legitimate DSM data stream?

(3) If not, what is a signal sequence that brings the receiver to the particular state. Compare pin level transition with working case.

kasius75, 2019-05-01:
hello,
My board is an HGLRC ZEUS and my receiver is a 4649T on UART6 TX and this is my experience.
If I plug first the board and then the battery, nothing moves in receiver tab and the board is in failsafe.
If I plug first the battery, most of the time, the board goes into failsafe. I must unplug and plug again several times (3 or 4) in order to get a normal boot sequence. Then every thing is OK and I can fly normally. And in this case, in the configurator, everything is OK in the receiver tab and the sticks move like expected.

jflyper, 2019-05-08:
@kasius75 A couple of questions for you.

1) What target firmware does your board use?

2) Is the issue specific to 4.0? If so, what is the last version that is known to work?

kasius75, 2019-05-08:
> @kasius75 A couple of questions for you.
> 
>     1. What target firmware does your board use?
> 
>     2. Is the issue specific to 4.0? If so, what is the last version that is known to work?

1) It's omnibusf4
2) Yes it's specific to 4.0.x, I also tried with 4.0.2. The last working version is 3.5.7.

kasius75, 2019-05-08:
> @kasius75 One more thing. Is 4649T in SRXL mode? That is, is it connected to TX6?
> 
> Better yet, CLI `diff` from working and non-working setup would be helpful.

Yes it's SRXL connected to TX6. Here is my diff:

# diff

# version
# Betaflight / OMNIBUSF4 (OBF4) 4.0.2 May  5 2019 / 12:07:47 (56bdc8d26) MSP API: 1.41

# start the command batch
batch start

board_name OMNIBUSF4
manufacturer_id AIRB

# name

# resources
resource PPM 1 NONE
resource SERIAL_TX 11 B14

# timer

# dma

# mixer
mixer CUSTOM
mmix 0  1.000 -0.963  0.774 -1.000
mmix 1  1.000 -1.000 -0.774  1.000
mmix 2  1.000  0.963  0.774  1.000
mmix 3  1.000  1.000 -0.774 -1.000

# servo

# servo mix

# feature
feature -AIRMODE
feature -RX_SPI
feature RX_SERIAL
feature MOTOR_STOP
feature SOFTSERIAL
feature TELEMETRY
feature LED_STRIP

# beeper
beeper -ARMED
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 2 16384 115200 57600 0 115200
serial 5 64 115200 57600 0 115200
serial 30 8192 115200 57600 0 115200

# led
led 0 0,0:SU:CT:8
led 1 0,1:SD:CT:8

# color

# mode_color

# aux
aux 0 0 4 1300 2100 0 0
aux 1 1 0 1750 2100 1 0
aux 2 1 2 900 1250 0 0
aux 3 2 0 1300 1700 1 0
aux 4 27 2 900 1250 0 0
aux 5 13 2 900 1700 1 0
aux 6 15 1 1500 2100 1 0
aux 7 28 4 1650 2100 0 0
aux 8 33 3 1750 2100 0 0
aux 9 34 3 900 1250 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 5 s 1100
rxfail 6 s 1100

# display_name

# master
set gyro_lowpass2_type = BIQUAD
set gyro_lowpass2_hz = 0
set dyn_notch_width_percent = 0
set dyn_lpf_gyro_max_hz = 650
set acc_calibration = -97,31,-35
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1100
set rssi_channel = 12
set serialrx_provider = SRXL
set blackbox_device = NONE
set max_throttle = 1950
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 30
set failsafe_off_delay = 100
set failsafe_throttle = 1195
set failsafe_procedure = AUTO-LAND
set vbat_scale = 112
set auto_disarm_delay = 10
set deadband = 3
set pid_process_denom = 1
set osd_vbat_pos = 14744
set osd_rssi_pos = 14688
set osd_tim_2_pos = 14720
set osd_avg_cell_voltage_pos = 14712
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_band = 1
set vtx_channel = 8
set vtx_power = 3
set vtx_freq = 5725

# profile
profile 0

set dterm_lowpass2_type = BIQUAD
set dterm_lowpass2_hz = 150
set feedforward_transition = 30
set iterm_rotation = ON
set smart_feedforward = ON
set f_pitch = 100
set f_roll = 100

# rateprofile
rateprofile 0

set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 195
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 50

# end the command batch
batch end

kasius75, 2019-05-08:
> @kasius75 One more thing. Is 4649T in SRXL mode? That is, is it connected to TX6?
> 
> Better yet, CLI `diff` from working and non-working setup would be helpful.

@jflyper You mean you want diff of 4.0 and 3.5?

mikeller, 2019-05-08:
@kasius75: Go into CLI, type `diff`, copy / paste the output into here, enclosed in 3 backticks (`).

kasius75, 2019-05-09:
@mikeller Yes, the 4.0 diff is 3 posts upper. But do you want 3.5 diff too?

mikeller, 2019-05-09:
Yes please.

kasius75, 2019-05-09:
Hello

I found a workaround to my problem. My aux3 switch is assigned to failsafe and beeper like below:

[image]

If I change to this config, everything is OK:

[image]

In this case, I don't have failsafe issue when the lipo is plugged. Also everything is OK in the receiver panel.

Any idea why the first configuration worked before 4.0 and not after?

closed 2026-09-26 by blckmn