betaflight/betaflight #8095: gps rescue toggle causes post flight stats to appear

erhmeh, 2019-04-25:
I needed to enable gps rescue today due to video loss. Unfortunately, this created some unexpected behavior. When I enabled it manually (not by failsafe), the post flight stats screen appeared (I had not disarmed, had 99LQ with xf), and the quad dropped for a bit (at 00:24) before gps_rescue kicked in and brought the quad back to where to took off from (as expected). However, as the video cleared up, I attempted to take it out of gps_rescue and it didn't want to change modes (still had 99LQ). Post flight stat screen stayed up until a few seconds before it hit the ground. Video of it at the below link:

https://youtu.be/WRMGm0jW0Eo

Sorry not going to try to reproduce this one for obvious reasons, almost had a heart attack first time. If you want to try, have a look at my settings in diff, put them on a quad and try and reproduce by changing from air mod to angle (was bound to switch's middle pos) to gps rescue.

Diff dump:
```
# version
# Betaflight / OMNIBUSF4FW (OBFW) 4.0.0 Apr 11 2019 / 07:30:57 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

board_name OMNIBUSF4FW
manufacturer_id 

# name
name JayFPV

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature GPS
feature TELEMETRY

# beeper

# beacon

# map

# serial
serial 2 2 115200 57600 0 115200
serial 3 0 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1500 2100 0 0
aux 1 1 1 1300 1700 0 0
aux 2 46 1 1800 2100 0 0
aux 3 35 2 1550 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_lowpass2_type = PT1
set gyro_lowpass2_hz = 100
set gyro_to_use = SECOND
set acc_calibration = 165,-76,-195
set rssi_channel = 8
set fpv_mix_degrees = 30
set serialrx_provider = CRSF
set dshot_idle_value = 450
set motor_pwm_protocol = DSHOT600
set failsafe_procedure = GPS-RESCUE
set align_board_pitch = 180
set ibata_scale = 150
set ibata_offset = 0
set small_angle = 180
set gps_provider = UBLOX
set gps_rescue_angle = 45
set gps_rescue_initial_alt = 70
set gps_rescue_descent_dist = 120
set gps_rescue_ground_speed = 1500
set gps_rescue_min_sats = 7
set gps_rescue_min_dth = 75
set gps_rescue_allow_arming_without_fix = ON
set pid_process_denom = 1
set osd_cap_alarm = 1300
set osd_alt_alarm = 150
set osd_tim1 = 2561
set osd_vbat_pos = 6156
set osd_rssi_pos = 6146
set osd_tim_1_pos = 2070
set osd_throttle_pos = 2434
set osd_crosshairs_pos = 6349
set osd_current_pos = 2454
set osd_mah_drawn_pos = 2136
set osd_craft_name_pos = 6199
set osd_gps_speed_pos = 2274
set osd_gps_lon_pos = 2402
set osd_gps_lat_pos = 2416
set osd_gps_sats_pos = 2053
set osd_home_dir_pos = 2222
set osd_home_dist_pos = 2243
set osd_altitude_pos = 2166
set osd_warnings_pos = 14728
set osd_avg_cell_voltage_pos = 6188
set osd_stat_max_dist = ON
set osd_stat_min_batt = OFF
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set osd_stat_flight_dist = ON
set vcd_video_system = NTSC

# profile
profile 1

set dterm_lowpass2_type = PT1
set dterm_lowpass2_hz = 100
set anti_gravity_gain = 10000
set feedforward_transition = 15
set p_pitch = 42
set i_pitch = 78
set d_pitch = 40
set f_pitch = 100
set i_roll = 68
set d_roll = 40
set f_roll = 100
set p_yaw = 40
set f_yaw = 100

# rateprofile
rateprofile 0

set roll_rc_rate = 107
set pitch_rc_rate = 107
set yaw_rc_rate = 102
set roll_srate = 82
set pitch_srate = 82
set yaw_srate = 78
```

Relevant Hardware:
 - FC: Airbot Omnibus F4 V6
 - GPS: Dual BN-220 GPS, Mounted on top of the battery for good signal
 - Control: XF TX, XF Nano RX, Immortal-T, Taranis QX7 (Not hardware modded, only software modded through lower baud rate)

atomiclama, 2019-04-25:
@erhmeh You  beat me to it, I had a similar experience yesterday whilst testing GPS rescue.  I have been running through different possibilities before submitting an issue, but seeing as you have entered this I'll add my setup.

I have GPS rescue setup on a switch and was testing via this method, I had a few successful test runs but the last one just showed up the post flight summary indicating a disarm and the quad dropped from the sky, but on the way down there was what looked like a rearm but it was too late. Sorry no FPV record.

GPS and RX were solid, I'm pretty sure I was over 100m out, well it was at a similar spot where a previous test had worked ok.

`Entering CLI Mode, type 'exit' to return, or 'help'

# diff all

# version
# Betaflight / OMNIBUSF4SD (OBSD) 4.0.0 Mar 14 2019 / 11:33:21 (360afd96d) MSP API: 1.41

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name OMNIBUSF4SD
manufacturer_id 
mcu_id 0036002c3037470f31313731
signature 

# name

# resources
resource MOTOR 2 A03
resource MOTOR 3 B01

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature GPS

# beeper

# beacon

# map

# serial
serial 0 64 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1750 2100 0 0
aux 1 1 5 1750 2100 0 0
aux 2 46 0 1750 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 4 h
rxfail 5 h

# display_name

# master
set gyro_lowpass_type = PT1
set dyn_notch_range = LOW
set dyn_notch_min_hz = 100
set dyn_lpf_gyro_min_hz = 70
set dyn_lpf_gyro_max_hz = 510
set acc_calibration = -12,92,243
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 14
set serialrx_provider = IBUS
set blackbox_p_ratio = 64
set dshot_idle_value = 600
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 5
set align_board_pitch = 180
set vbat_min_cell_voltage = 310
set vbat_warning_cell_voltage = 330
set vbat_scale = 111
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_allow_arming_without_fix = ON
set deadband = 5
set yaw_deadband = 5
set pid_process_denom = 1
set osd_warn_crash_flip = OFF
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_tim1 = 2561
set osd_rssi_pos = 2084
set osd_link_quality_pos = 43
set osd_tim_1_pos = 2260
set osd_ah_pos = 2126
set osd_current_pos = 2274
set osd_mah_drawn_pos = 2293
set osd_gps_speed_pos = 2242
set osd_gps_lon_pos = 2402
set osd_gps_lat_pos = 2415
set osd_gps_sats_pos = 2098
set osd_home_dir_pos = 2348
set osd_home_dist_pos = 2350
set osd_altitude_pos = 2178
set osd_avg_cell_voltage_pos = 2306
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_flight_dist = ON
set debug_mode = RC_SMOOTHING
set vcd_video_system = NTSC

# profile
profile 0

set dyn_lpf_dterm_min_hz = 70
set dyn_lpf_dterm_max_hz = 150
set dterm_lowpass_type = PT1
set dterm_lowpass2_hz = 0
set vbat_pid_gain = ON
set anti_gravity_gain = 10000
set p_pitch = 35
set d_pitch = 25
set f_pitch = 32
set p_roll = 33
set d_roll = 23
set f_roll = 30
set p_yaw = 30
set i_yaw = 90
set d_min_roll = 0
set d_min_pitch = 0

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save`

Black box log which shows no disarm state .

[image]

[LOG00018.zip](https://github.com/betaflight/betaflight/files/3115864/LOG00018.zip)

etracer65, 2019-04-25:
@erhmeh If you were closer than 100m then GPS Rescue will disarm when activated. Also you have `gps_rescue_allow_arming_without_fix = ON` so if you took off without having enough satellites and establishing a home position fix then GPS Rescue cannot work and will disarm. Lastly if you don't have enough satellites (you have the limit set to 7) then it will disarm. Read the documentation:

https://github.com/betaflight/betaflight/wiki/GPS-rescue-mode

@atomiclama Same comments. You were likely too close on the last activation of GPS Rescue. Also there won't be any "disarm" in the log as that shows the arming switch, not the general armed/disarmed state.

atomiclama, 2019-04-25:
@etracer65 Didn't know that about the arm in the log cheers, as to the rest, num sats home position etc were all ok and a previous rescue worked. Probably the distance thing but can't be sure.

erhmeh, 2019-05-01:
@atomiclama Maybe we should reconsider what the expected behavior should be- in my case I was standing under a tree and enabled gps rescue to take over control while I took a few steps forward to regain video signal before taking back control. I think under all circumstances, the pilot should be allowed to retake control of their quad if possible.

etracer65, 2019-05-01:
Then I suggest you submit a pull request to add "pilot under a tree" safety detection so that GPS Rescue activation can occur without the minimum distance limit.

closed 2019-06-07 by stale[bot]