betaflight/betaflight #8096: OSD Profile

denver1370, 2019-04-26:
I would like to request for improvement on the OSD profile feature that got added recently.
The OSD profile I believe only stores flags of items which are enabled.
So I think it is also a good feature to store the coordinates of the items being displayed in the OSD.

For example I want to change my OSD profile from my Freestyle to Long Range. I may wanted to have same items enabled but will be placed in different position.

I hope it won't affect the firmware size too much.

RipperDrone, 2019-04-26:
wow, tripling the x/y data sets shouldn't be a small burden indeed... sometimes I feel we are going overboard with aethetics refinements rather than focusing on best flight performance which is core to BF claim. I'd rather put my vote in for dedicating some space to extending smart filters or battery cell auto-switch profiles (4S <--> 6S: adjust PIDs / filters by scaling etc.) rather than further loading the OSD which is already overwhelming regarding the new features (motos stats, debug options, ...). But let's hear others as well...

denver1370, 2019-04-27:
> wow, tripling the x/y data sets shouldn't be a small burden indeed... sometimes I feel we are going overboard with aethetics refinements rather than focusing on best flight performance which is core [...]

Yes indeed this feature should be consider as least priority of course compare to features that can help for performance enhancement.
Maybe I should rephrase what I just said from "I would like to request" to "Is it possible to request
", (sorry my bad). 
But I don't know, because I cannot also tell how much effort it needs and how will it affect the firmware size. This is just to share the idea that come in my mind. :)

pkruger, 2019-04-27:
This was discussed when the OSD Profiles were implemented.  I like the idea, but will have a significant impact on code size/storage for each element as mentioned above.  Also, having elements in different locations on different profiles makes it hard to find what you are looking for because you get use to where something is.  That said, sometimes you just want something below something else, i.e. in the same area, but a little differently arranged however that doesn't really justify the impact this change will have on code size/storage for OSD elements.  (Some features have already been removed for F3 flight controllers.)

Docteh, 2019-04-27:
For long range have you looked at inav? It's got separate osd profiles and a GPS position hold. I think we should try and be careful with remaining code space on current hardware. Unified build for STM32F411 is already 85.29%. STM32F7X2 (722) is in a similar position.

denver1370, 2019-05-17:
Thank you guys for the effort for even checking this request.
I think this case can be close as it was totally not needed.

closed 2019-05-17 by denver1370