betaflight/betaflight #8098: Create a LUA script as "yaapu/FrskyTelemetryScript" or "Inav"

backman8571, 2019-04-26:
Create a LUA script as "yaapu/FrskyTelemetryScript" or "Inav"

etracer65, 2019-04-26:
What exactly is your question or issue?

backman8571, 2019-04-26:
I like to see this type of view

[image: sample]
<https://github.com/iNavFlight/LuaTelemetry/blob/master/assets/iNavQX7.png>
   [image: sample]
<https://github.com/iNavFlight/LuaTelemetry/blob/master/assets/iNavX9D.png>
<https://github.com/iNavFlight/LuaTelemetry#pilot-glass-cockpit-view-for-fixed-wing-pilots>Pilot
(glass cockpit) view for fixed wing pilots

[image: sample]
<https://github.com/iNavFlight/LuaTelemetry/blob/master/assets/iNavQX7pilot.png>
   [image: sample]
<https://github.com/iNavFlight/LuaTelemetry/blob/master/assets/iNavX9Dpilot.png>
<https://github.com/iNavFlight/LuaTelemetry#radar-map-view>Radar (map) view

[image: sample]
<https://github.com/iNavFlight/LuaTelemetry/blob/master/assets/iNavQX7radar.png>
   [image: sample]
<https://github.com/iNavFlight/LuaTelemetry/blob/master/assets/iNavX9Dradar.png>
<https://github.com/iNavFlight/LuaTelemetry#altitude-graph-view>Altitude
graph view

[image: sample]
<https://github.com/iNavFlight/LuaTelemetry/blob/master/assets/iNavQX7alt.png>
   [image: sample]
<https://github.com/iNavFlight/LuaTelemetry/blob/master/assets/iNavX9Dalt.png>
<https://github.com/iNavFlight/LuaTelemetry#horus-view>Horus view

[image: sample]
<https://github.com/iNavFlight/LuaTelemetry/blob/master/assets/iNavHorus.png>

Thats what i can see on my Hours/T16 if i had an aurdino

Can we get the same for BetaFlight
Regards

Per Backman

Systems Management/Network Management
Professional Services

CELL : 4698558729

Email1 : per.backman@aurea.com
Email2 : per.backman@zephyrtel.com
SKYPE: backman8571@live.com

etracer65, 2019-04-26:
These are just telemetry Lua scripts. Somebody could create these if they wanted. Looks like all the necessary data is available in the telemetry sensors so I don't really see how this is a Betaflight issue. So feel free to implement these scripts on your radio if you want.

backman8571, 2019-04-26:
But this script will not work with beta flight using a different path to
send data

That's why it's a betaflight issue/new function

etracer65, 2019-04-26:
The point is that this wouldn't be something we'd create. More likely to be created by the community if someone was interested in pursuing.

backman8571, 2019-04-26:
So can you put out and see if anyone is interested doing that ??

mikeller, 2019-04-28:
Closing, as this is not a firmware issue, and there is no project to move this to.

closed 2019-04-28 by mikeller