betaflight/betaflight #8101: Release Candidate 6: Smart Audio 2.1 now working

LikeCatcher, 2019-04-26:
**Describe the bug**
BetaFlight 4 Release Candidate 6: **Smart Audio 2.1 now working with TBS Unify Race 2 VTX**.
_When VTX was swapped out to a TBS Unify Pro 5G8 HV the Smart Audio functioned correctly._

**To Reproduce**
1. Install Firmware on wlight controller.
# version
# Betaflight / PYRODRONEF4 (PYRO) 4.0.0 Apr  3 2019 / 14:20:12 (22b9f3453) MSP API: 1.41 
2. Enable UART5 | Peripherals | enable VTX(TBS Smart Audio) AUTO
3. Save and Reboot
4. Disconnect from BetaFlight
5. Connect Battery to PYRO32F3-45Amp 4in1 F3 32bit BLHELI ESC 4x45A.
6. Using Fr-Sky X-Lite and FPV gogles, Move sticks to enter OSD controll Main Menu.
7. Scroll down to PROFILE
8. FEATURES
9. VTX SA screen displays VTX SMART AUDIO ------------------- >STATX  BACK
10. VTX TR screen displays full Tramp controlls that can be set but have no effect.

**Expected behavior**
VTX SA tab would function normally.
VTX TR tab would not function.

**Flight controller configuration**
# diff

# version
# Betaflight / PYRODRONEF4 (PYRO) 4.0.0 Apr  3 2019 / 14:20:12 (22b9f3453) MSP API: 1.41

# start the command batch
batch start

board_name PYRODRONEF4
manufacturer_id 

# name
name Joel

# resources
resource PINIO 1 A15

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature ESC_SENSOR

# beeper

# beacon

# map

# serial
serial 0 64 115200 57600 0 115200
serial 1 1024 115200 57600 0 115200
serial 2 32 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 900 1200 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 2 900 1300 0 0
aux 4 19 4 900 1300 0 0
aux 5 35 4 1700 2100 0 0
aux 6 36 3 900 1300 0 0
aux 7 39 6 900 1850 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_lowpass2_type = PT1
set gyro_lowpass2_hz = 100
set acc_calibration = -6,-2,-19
set serialrx_provider = SBUS
set motor_pwm_protocol = DSHOT1200
set align_board_pitch = 180
set align_board_yaw = 180
set bat_capacity = 1300
set yaw_motors_reversed = ON
set pid_process_denom = 1
set tlm_halfduplex = OFF
set osd_units = IMPERIAL
set osd_vbat_pos = 2296
set osd_rssi_pos = 2272
set osd_tim_2_pos = 2456
set osd_flymode_pos = 2074
set osd_vtx_channel_pos = 2048
set osd_craft_name_pos = 2062
set osd_warnings_pos = 14729
set osd_avg_cell_voltage_pos = 2328
set osd_disarmed_pos = 2282
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set vtx_band = 1
set vtx_freq = 5865
set vcd_video_system = NTSC
set pinio_box = 39,0,0,0

# profile
profile 0

set dterm_lowpass2_type = PT1
set dterm_lowpass2_hz = 100

# rateprofile
rateprofile 0

# end the command batch
batch end

**Setup / Versions**
HYPERLITE F4 OSD V2 WITH VTX PIT MODE purchased from https://pyrodrone.com/collections/flight-controllers/products/hyperlite-f4-osd

PYRO32F3-45AMP 4IN1 F3 32BIT BLHELI ESC 4X45A https://pyrodrone.com/collections/electronic-speed-controllers/products/pyro32-45amp-4in1-by-pyrodrone

**Build List:**
Catalyst Machineworks Raging Droner 5" frame
Lumenier MX2206-9 2450Kv POPO motors

PyroDrone F4 flight controller
PYRO32F3-45Amp 4in1 F3 32bit ESC (with attached 2 supplied caps)
_---ESC and FC are wired by supplied connector_

TBS Unify Pro Race2 MMCX
_---wired to |VID|VBAT|GND|TX5 (Smart Audio)|_

FrSky receiver R-XSR
_---wired to |SBUS|5V|GND|TELE (S.port)|_

Hellgate Buzzer
_---wired to |GND (cam gnd)|BUZ+|BUZ-|_

RunCam MicroEagle
_---wired to |CAM_S|VBAT|GND|_

 - how are the different components wired up.
See above.

**Additional context**
Beginning of my YouTube video review talks about these issues https://youtu.be/zakNVOaSy00

etracer65, 2019-04-26:
So you opened an issue to tell us something is working? And why use RC6 when the 4.0 has been officially released.

LikeCatcher, 2019-04-27:
No, just the opposite. It is not working. RC6 does not fix the problem.
Thanks,

Joel Lukacher
LikeCatcher
| Social Media
| Email Marketing
| Responsive Websites
| Reputation Management
| Streaming Video & Photo
| Drone Video & Photo
754-200-1884
joel@likecatcher.com
www.LikeCatcher.com

<author@noreply.github.com>
Sent: 4/26/2019 7:48:12 PM
2.1 now working (#8101)

LikeCatcher, 2019-04-27:
No, just the opposite. It is not working. RC6 does not fix the problem, 
or the 4.0 release. TBS instructed me to load RC6 to fix the Smart Audio 
v2.1 because it does not work in the 4.0 release.  Sorry if I didn't 
make that clear.
Thanks,

Joel Lukacher
LikeCatcher
| Social Media
| Email Marketing
| Responsive Websites
| Reputation Management
| Streaming Video & Photo
| Drone Video & Photo
754-200-1884
joel@likecatcher.com
www.LikeCatcher.com

etracer65, 2019-04-27:
Sorry, but you're not making it any clearer.

Subject of the Issue: "Release Candidate 6: Smart Audio 2.1 now working"

Then you say:
> It is not working. RC6 does not fix the problem

LikeCatcher, 2019-04-27:
Please re-read my original report above. I corrected my typos that made it unclear. Sorry.

LikeCatcher, 2019-04-27:
I'was a little nervious writing my first report . Please excuse my faux pas. All is correct now.

LikeCatcher, 2019-04-27:
I corrected my report to clarify. I really want to be helpful. Do you 
have any other questions?
Thanks,

Joel Lukacher

Sent: 4/26/2019 8:35:39 PM
2.1 now working (#8101)

LikeCatcher, 2019-04-27:
I'm learning this Github system on the fly. My 1st time here. Your understanding is appreciated.

watageek, 2019-04-27:
@LikeCatcher

I am not a dev nor someone who knows what he is talking about. There is so much combination possible with available hardware. The devs and maintainers are mere mortals afterall and the hobby is deeply incrusted ( somewhere between addiction and obsession).  Have a look on the jenkins. Do a dissect ( i.e. go back in "time" to find the builds that works, ex g: build 1470 works and not the 1471).  Help them to find where is the glitch providing crunchy material. 

( hope not to be roasted)

watageek, 2019-04-27:
https://ci.betaflight.tech/job/Betaflight/

matbhe, 2019-05-05:
#7979

RyanHolcomb, 2019-05-10:
I have the same issue with the TBS Evo vtx.
SA 2.1 does bot work on BF 4.0. My OSD > VTX SA only displays StatX and BACK. 

Using the same wire harness I unplugged my Evo vtx from my quadcopter build and plugged in my Unify Pro HV running SA 2.0 and that works. That eliminates my wires, soldering or menu navigation as the problem. It seems to be either BF 4.0 or the TBS Evo.

denver1370, 2019-05-17:
I have same issue with my TBS Unify Pro Nano.
Sometimes I got the SA properly working and sometimes it is not.

I think there are two separate firmwares that may need to look at.
1. TBS firmware - this is my first TBS vtx so, sorry I cannot test and compare with other VTX running SA
2. Betaflight firmware

Cases found:
a. If SA is available in OSD - After doing Save and Reboot. it will disappear at reboot
b. If SA is not available in OSD. Going to VTX TR, changing and setting some stuff then Save and Reboot will have the SA available again.

Hope this helps. will update again after finding out what causes this.
Also considering the 3.3V requirements for the Nano VTX signal.
Im running Mambo F405 FC. 
betaflight 4.0.2

badjeff, 2019-06-17:
I have same issue with my TBS Unify Pro32 HV. SA 2.1 does not work on BF 4.0.x and 4.1. My VTX SA OSD only show STATX and BACK. The BADLEN counter in STATX increase continuously. After digging the source code and SA v2.1 spec. I figured out the length of GET SETTING command response from Unify Pro32 HV is slightly larger than the buffer size of response frame. After increasing the value of constant SA_MAX_RCVLEN, compile, flash, and set vtxtable values in CLI, It is is able to change Bands and Chanels via OSD.
[ Ref: https://www.team-blacksheep.com/tbs_smartaudio_rev09.pdf ]
[ Pull Request: https://github.com/betaflight/betaflight/pull/8438 ]

DeezyFPV, 2019-06-18:
Same issue as you badjeff, using a pro32 and smart audio is only showing statx

mantasnakt, 2019-06-20:
I can confirm that TBS Unify Pro32 HV. SA 2.1 does not work on BF 4.0.x . Only STATX and BACK in BF OSD.

01ivr3, 2019-06-21:
I was able to get SA working on my "TBS Unify Pro32 HV" using the latest 4.1.0 #1706 Dev release and SA v2.1 table specified in #8276, thanks to help from @fujin on Slack 👍

closed 2019-07-28 by stale[bot]

pr0gg3r, 2019-08-18:
how can I set the vtxtable?

patrick131184, 2019-08-20:
https://github.com/betaflight/betaflight/pull/8276

viktorglekler, 2019-10-29:
I also have the problem with smartaudio 2.1 and betaflight. My vtx is tbs pro hv race 2 and betaflight version is 4.1. I tried different uart ports, but it doesnt work..