betaflight/betaflight #8104: Using restore from json causes rates configuration issue on pitch RC rate

eplewis89, 2019-04-27:
Restored settings to a BF4.0.0 firmware board, after upgrading from 3.5.7. This caused an issue with the pitch RC rate to not be populated (it was 0.0) whereas on the previous version, it use the roll RC rate. This causes the velocity to be "NaN", and the quad is not controllable on the pitch axis

**To Reproduce**
Steps to reproduce the behavior:
1. Start with a 3.5.7 board
2. Backup settings to json
3. Upgrade 3.5.7 board to 4.0.0
4. Restore settings from json

**Expected behavior**
RC rate for pitch will be populated with previous value from 3.5.7

This was using a JB F4 AIO v2

etracer65, 2019-04-27:
Don’t try to restore settings from an earlier version. This is not supported.

eplewis89, 2019-04-27:
@etracer65 aside from pitch RC rate and osd settings, everything else worked properly

etracer65, 2019-04-27:
So you think. Don’t do this, you’ll have problems.

closed 2019-04-27 by eplewis89

etracer65, 2019-04-27:
The problem is that you’ve restored settings from 3.5 that may have new defaults in 4.0 - particularly PID and filtering settings that have changed dramatically. Don’t restore settings from 3.5 or you will have problems.