betaflight/betaflight #8105: Betaflight 4.0 (OMNIBUSF4FW) MOTORS_STOP option not working

simtcrom, 2019-04-27:
**Describe the bug**
I have updated my 88 micro brushless quad (Airbot Omnibus F4 Nano V6, OMNIBUSF4FW) to betaflight 4.0.

One issue I found.
Irrespective of MOTOR_STOP enabled or not, when I arm the quad, motors start spinning.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
Enabling MOTOR_STOP function should not spin the motors when quad is armed.

**Flight controller configuration**
Airbot Omnibus F4 Nano V6, OMNIBUSF4FW

jflyper, 2019-04-27:
Config `diff` please.

simtcrom, 2019-04-27:
> Config `diff` please.

[diff_OMNIBUSF4FW_4.0.txt](https://github.com/betaflight/betaflight/files/3123656/diff_OMNIBUSF4FW_4.0.txt)

Asizon, 2019-04-27:
I think that with AIRMODE feature enabled, engine_stop will not respond to commands

simtcrom, 2019-04-27:
> I think that with AIRMODE feature enabled, motor_stop will not respond to commands. Now airmode is enabled by default,try dissabling it and test

that fixed it. thank you.

simtcrom, 2019-04-27:
AIRMODE feature enabled, motor_stop will not respond to commands. Now airmode is enabled by default,try dissabling it and test

closed 2019-04-27 by simtcrom