betaflight/betaflight #8107: OSD Profile Names

pkruger, 2019-04-27:
**Is your feature request related to a problem? Please describe.**
When different pilots on a team use the same quad, they have to remember which profile to use, 1, 2 or 3.  Also it is not clear from the DVR which pilot was flying.  

**Describe the solution you'd like**
If a name for each OSD profile was added as well as an OSD profile name OSD element, each pilot can select the OSD profile by their name being displayed in the OSD.  Also, this will also be very handy if PID and rate profiles are on the same switch as the OSD Profile selection.  Each pilot just selects their OSD profile and their PID and rate profiles are selected automatically - they don't have to remember which profile number they are.

**Describe alternatives you've considered**
Using profile numbers, but this does not solve the issue of knowing which pilot was flying in      the DVR.

pkruger, 2019-05-07:
@mikeller I'm about to start with this.  Do you want to wrap this in #define USE_OSD_PROFILE_NAMES, I think USE_OSD_PROFILES is enough?

mikeller, 2019-05-07:
Making this conditional to `USE_OSD_PROFILES` should be fine.

pkruger, 2019-05-16:
PR: https://github.com/betaflight/betaflight/pull/8272

PR #8272 (2019-05-25): Add OSD profile names feature
Fixes: #8107
closed 2019-05-25 by mikeller