betaflight/betaflight #8109: Buzzer not working since BetaFlight 4.0

ChDUP, 2019-04-27:
Setup : ImmersionRC Vortex 230 Mojo stock
Configurator: 10.5.1
Firmware: BTFL 4.0.0
Target: IRCSYNERGYF3

I have a problem with my Mojo, since I've upgrade to BF4
I had no problem with 3.2 version.
Buzzer is not working at all 
not with switch, not with failsafe, not with powered on ...

here is my configuration :

> gyro_hardware_lpf = NORMAL
> Allowed values: NORMAL, 1KHZ_SAMPLING
> 
> gyro_sync_denom = 1
> Allowed range: 1 - 32
> 
> gyro_lowpass_type = BIQUAD
> Allowed values: PT1, BIQUAD
> 
> gyro_lowpass_hz = 150
> Allowed range: 0 - 16000
> 
> gyro_lowpass2_type = BIQUAD
> Allowed values: PT1, BIQUAD
> 
> gyro_lowpass2_hz = 0
> Allowed range: 0 - 16000
> 
> gyro_notch1_hz = 0
> Allowed range: 0 - 16000
> 
> gyro_notch1_cutoff = 0
> Allowed range: 0 - 16000
> 
> gyro_notch2_hz = 0
> Allowed range: 0 - 16000
> 
> gyro_notch2_cutoff = 0
> Allowed range: 0 - 16000
> 
> gyro_calib_duration = 125
> Allowed range: 50 - 3000
> 
> gyro_calib_noise_limit = 48
> Allowed range: 0 - 200
> 
> gyro_offset_yaw = 0
> Allowed range: -1000 - 1000
> 
> gyro_overflow_detect = ALL
> Allowed values: OFF, YAW, ALL
> 
> yaw_spin_recovery = ON
> Allowed values: OFF, ON
> 
> yaw_spin_threshold = 1950
> Allowed range: 500 - 1950
> 
> dyn_notch_range = AUTO
> Allowed values: HIGH, MEDIUM, LOW, AUTO
> 
> dyn_notch_width_percent = 8
> Allowed range: 0 - 20
> 
> dyn_notch_q = 120
> Allowed range: 1 - 1000
> 
> dyn_notch_min_hz = 150
[... 916 lines cut ...]
> rcdevice_feature = 0
> Allowed range: 0 - 65535
> 
> gyro_1_bustype = SPI
> Allowed values: NONE, I2C, SPI, SLAVE
> 
> gyro_1_spibus = 2
> Allowed range: 0 - 3
> 
> gyro_1_i2cBus = 0
> Allowed range: 0 - 2
> 
> gyro_1_i2c_address = 0
> Allowed range: 0 - 119
> 
> gyro_1_sensor_align = CW270
> Allowed values: DEFAULT, CW0, CW90, CW180, CW270, CW0FLIP, CW90FLIP, CW180FLIP, CW270FLIP

I've read it could exist problem with active/passive buzzer.
I haven't find if buzzer on the mojo is active or passive.
But I would like to try change the frequency but when I try on the CLI :
`set beeper_frequency = 0`
it returns : `###ERROR: INVALID NAME###`

Thanks for your help

etracer65, 2019-04-27:
Prior to 3.5 there was no Betaflight target for this board. IRC was supplying a custom firmware. When the target was created in 3.5 it was set up without any beeper support. So it would take modifications to the target and documentation from the vendor to identify which pin the beeper is connected to on the MCU (and whether it's active/passive and inverted or not).

ChDUP, 2019-04-28:
My previous version was 3.2.2
I think it was the BF installed in stock version of the mojo.

> actually it was added in 3.4.1. But the same issues still apply.
What do you mean ? beeper is also not working in 3.4.1 ?

ChDUP, 2019-04-28:
Ok
thanks for your response.
It could be possible by watching the custom firmware by IRC.
But they only deliver compiled .hex of the firmware. 
Anyone has another idea ?

mikeller, 2019-04-28:
@ChDUP: Yes. Raise this issue with IRC - they are the only ones that have the information required to fix this.

closed 2019-06-04 by stale[bot]