betaflight/betaflight #8110: Easy way to find drone without latitude & longitude

leisim, 2019-04-27:
Today I barely lost my drone because it lost signal and it dropped into tall grass. The battery ejected and I searched for three hours until I found it.

**Problem**
I don't have the GPS coordinates in my OSD because it takes too much space.

**Solution**
I heard from multiple pilots that the wasted space is not worth it for them.
I think the best solution is to make the location more compact in the video feed.
I propose to use Google's [open location code](https://github.com/google/open-location-code) (also known as "plus codes") instead of coordinates in the OSD.

I took my flying spot as sample:
GPS:           48,0800375, 11,5680156
plus code: 3HJ9+26

Plus codes have 6 characters for an accuracy of 14x14m or 7 characters for 3x3m. Any coordinate can be easily converted to a plus code.

**Implementation**
Since Google provides open source implementations for a variety of languages ([including C](https://github.com/google/open-location-code/blob/master/c/example.c)) the implementation of this feature would be rather simple. If you like the idea, I will create a pull request.

etracer65, 2019-04-28:
Duplicate. https://github.com/betaflight/betaflight/issues/7788

closed 2019-04-28 by etracer65

leisim, 2019-04-28:
Oh sorry :/