betaflight/betaflight #8118: Acro Trainer Mode is enabled opposite to expected

knightz4u, 2019-04-28:
Hi,

There may be a bug with the Acro Trainer mode activation.

I have tried to configure it so that it activates with two switches by adding a second range so that it uses two auxiliary channel switch positions (and used logical AND for both ranges), but once i did this, the mode color (orange in BF 4.0) shows it is deactivated when both switches are in the correct set ranges, and then when one is outside it shows the Acro Trainer is activated (changes color to Orange).

Can you verify you get the same? and what the implication of this is?

watageek, 2019-04-28:
post a diff/dump.

knightz4u, 2019-04-28:
here is the Diff and Dump output

# diff

# version
# Betaflight / LUXF4OSD (LUX4) 4.0.0 Apr 11 2019 / 07:24:15 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

board_name LUXF4OSD
manufacturer_id 

# name

# resources
resource MOTOR 1 A03
resource MOTOR 2 B00
resource MOTOR 3 A02
resource MOTOR 4 B01
resource LED_STRIP 1 NONE
resource SERIAL_TX 11 A01

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature SOFTSERIAL
feature GPS

# beeper
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon

# map

# serial
serial 0 64 115200 57600 0 115200
serial 5 2 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200
[... 1002 lines cut ...]
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 50
set tpa_breakpoint = 1500
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end

# 
[image]

knightz4u, 2019-04-28:
This only happens when having two ranges for Acro Trainer, with one range the operation is correct.

In the snapshot, the Acro Trainer mode should not be active, Betaflight shows it to be active.

In the next snapshot, the Acro Trainer mode should be active,but Betaflight shows it to be inactive.

Let me know your thoughts.

[image]

watageek, 2019-04-28:
Yap. Reproducible.

Two range gives NAND. Interesting. 😂

Docteh, 2019-04-28:
@knightz4u can you try the same thing with GPS rescue? ACRO TRAINER is mode 47, and GPS RESCUE is 46

I tried other modes like BEEP and CRASHFLIP and it seems to work. Just trying to narrow down the problem.

watageek, 2019-04-28:
@Docteh 
keen observation
I made a custom firmware without accro trainer. 
Launch control is the **last** mode in the tab.
[image]

knightz4u, 2019-04-28:
@Docteh sorry couldn't verify the GPS Rescue, it is not coming on because my GPS antenna has a problem and can't catch enough satellites. Will be a while till i receive the replacement i ordered.

Docteh, 2019-04-28:
@knightz4u oh I didn't mean actually go fly, I just meant try and light up the option in the configurator. I don't have a GPS on my drone, so I don't know exactly what is needed to let it show up as a mode. It seems like the modes after Paralyze (45) are having this trouble, but the issue isn't jumping out of the code for me.

Docteh, 2019-04-28:
Looks like in the boxId_e enumeration BOXPARALYZE is the 32nd item. 

https://github.com/betaflight/betaflight/blob/6a67972c6cade166741857daaf8695dbd0a96796/src/main/fc/rc_modes.h#L29-L78

I'll have to leave figuring out what the bitmask stuff is doing for now.

mikeller, 2019-04-28:
@Docteh: The boxes themselves should be designed to allow larger sets than 32, but it looks like there's s bug affecting the (new) 'logical and' functionality.

Docteh, 2019-04-30:
@knightz4u When 4.0.1 is out please give it a shot. Should be good to go.

PR #8130 (2019-05-01): Fix to make the AND Logic work on all modes
This fixes #8118
32 bit problems require 32 bit solutions.
commit 130266af (2019-05-01): Fix to make the AND Logic work on all modes

This fixes #8118
closed 2019-05-01 by mikeller