betaflight/betaflight #8122: Roll of Death each time I use full stick roll/pitch.

Kikerumbo, 2019-04-28:
Fellow pilots,

I've just maiden my two new builds, both 6S low kv. One it's an Armattan Marmotte with Flywoo 2306 1750kv, Airbot Furling 32 - 45A and a iFlight SucceX F7 TwinG (dual gyro). The other has Emax ECO 2306 1700kv, Diatone F405 Mamba (40A/F4) stack.

The one with the Mamba stack is flying ok, a little bit of vibrations, but that (I assume) is solvable with PID tuning.

The one with the F7 dual gyro and the Airbot Furling ESC is doing Rolls of Death each and every time I go full stick to roll or pitch. If I don't go to full deflection, the quad flies well.Both have the same configuration after trying both recommended setting in the Betaflight tuning tips:

The "I'm a freestyle pilot, what should I paste into the CLI?" with PIDs lowered one third and then the "I'm more of a race pilot... or 6S/lightweight responsive machines". Both, same results. The cheap quad flies with vibrations and the "expensive" one flies well, but with Rolls of Death every time I do a full roll/pitch stick.

By the way, deg/s are under 950.

Thanks for any help provided.

etracer65, 2019-04-28:
As requested when you created the issue, please provide a `diff` of your settings. Also a blackbox log showing the problem is required. Lastly this particular flight controller tends to be poorly designed and has an unusually high number of gyro failures. Try testing with each gyro independently. `set gyro_to_use = FIRST` and `set gyro_to_use = SECOND`.

Kikerumbo, 2019-04-28:
Thanks for your answer. I'm sorry for not providing the log and the diff.

I'll try what you propose and come back with the results.

Really sad about hearing that a +40€ flight controller from iFlight is crap. It's been released a few weeks ago and it's already pretty bad...

Kikerumbo, 2019-05-02:
I tried the new Betaflight version 4.0.1, but the result was the same.

I attach two blackbox logs, one with 5.1x4.6x3 HQ props and the second with 5x4.3x3 HQ. With the latter props it was harder to provoke the Roll of Death, but it happened nonetheless

The first one is with the stock Betaflight 4.0.1 and the next one is with the recommended settings by UAV TECH for 6S. Same result.

Still have to test the "set gyro_to_use = FIRST and set gyro_to_use = SECOND", but I wanted to try the new Betaflight version, because in the list of changes comes that some dual gyro issues were solved.

I'm a bit desperate, to be honest. If you could help, I'll appreciate it.

Have a good day.

[Marmotte_DIFF.txt](https://github.com/betaflight/betaflight/files/3139837/Marmotte_DIFF.txt)
[Blackbox_Logs.zip](https://github.com/betaflight/betaflight/files/3139843/Blackbox_Logs.zip)

Kikerumbo, 2019-05-03:
Confirmed by iFlight staff, one of the gyro is malfunctioning.

They are going to give me a refund, so... that's the end of the story.

Thanks for the help.

closed 2019-05-03 by wind0r