betaflight/betaflight #8123: OSD Display Watt Hours per Unit Distance

yujanshrestha, 2019-04-29:
**Is your feature request related to a problem? Please describe.**
No. It will be convenient to gauge the efficiency of a particular rig, independent of cell count and battery sag. 

**Describe the solution you'd like**
Only available if GPS and some type of current sensor are available. 

A new display on the OSD called "Watt hours per mile" or "Watt hours per km". This will be calculated by multiplying instantaneous power (watts)  / speed (miles / hour) =  watt * hours / mile

It would also be great if an average were displayed at the end of the flight statistics.

**Describe alternatives you've considered**
DVR freeze frame and estimating the metric after the flight. An instantaneous value will be useful.

**Additional context**
iNav seems to have a similar `OSD_EFFICIENCY_WH_PER_KM` in `osd.h` https://github.com/iNavFlight/inav/blob/master/src/main/io/osd.h#L78 . This seems to be exactly what I am looking for.

ATKCA, 2020-07-27:
Yes add this. Also expand with mAh per kilometre / mile and or perhaps kilometres / miles till empty.
We could even go all out and use the distance till empty figure and then minus distance to home to create "RTH warning" allowing the pilot to try to reduce throttle to achieve more efficiency to make it home.

daijo, 2024-07-04:
Just want to add that this still would be super useful! Use it on INAV and Betaflight needs this too.

realThirdpartycookie, 2025-02-01:
Still a necessary feature for us long range folk

haslinghuis, 2025-02-01:
Was this not implemented in https://github.com/betaflight/betaflight/pull/11843 ?

closed 2026-09-26 by blckmn