betaflight/betaflight #8128: BLHeli32 (almost) can't connect to ANYFCF7

andrejpodzimek, 2019-04-30:
BLHeli32 (both the Windows suite under Wine and the new native Linux suite) appear to time out when trying to connect to my `ANYFCF7` FC. It might well be a BLHeli issue, but beacause it's related to Betaflight versions, I thought I'd ask here first.

I'm geting the failure on Betaflight 4.0.0 with [this `diff all`](https://github.com/betaflight/betaflight/files/3129805/diff_all.txt) and [this `dump all`](https://github.com/betaflight/betaflight/files/3129807/dump_all.txt). Surprisingly, **about one attempt out of 50 succeeds** (hence the *almost* in the title). No idea why; there doesn't seem to be any pattern (tried with battery connected / disconnected, with / without prior Betaflight GUI connections etc, but spotted no difference). Once it's connected, I can configure ESCs normally. But it takes ~tens of minutes of failed attempts to ultimately connect.

BLHeli32 first takes obviously long to connect, then reboots the flight controller (which is announced by a pop-up window) and ultimately fails to (re)connect nonetheless:
[image]

BLHeli32 access works for me in all other circumstances:
* the same FC with Betaflight 3.5.7
* various other FCs with Betaflight 4.0.0 (a number of F3s and F4s at least)

Despite the BLHeli32 configurator issue, the quad flies just fine. ESCs work. ESC telemetry works. It uses DShot600.

I tried to use the `STM32F745` target (from `master`) instead of `ANYFCF7`, following [the `hardware dump` howto](https://github.com/betaflight/betaflight/blob/master/unified_targets/docs/CreatingAUnifiedTarget.md). Everything seemed to work fine, hardware was recoginzed, but the BLHeli32 problem was still there.

An observation: There is a strange delay of ~2 seconds when the GUI establishes a connection to this particular `ANYFCF7` with Betaflight 4.0.0. I don't see that in any other cases; normally the GUI connects instantaneously, AFAICT. So I'd hypothesize that if such a delay occurs also when BLHeli32 tries to connect, it might hit a timeout, occasionally succeeding by a narrow margin. Not sure if the timeout is tunable though; I couldn't find that anywhere.

mikeller, 2019-04-30:
@andrejpodzimek: I suspect this is another incarnation of the issue raised in #7873: The STM libraries currently used in Betaflight for F7 have issues in the USB stack. We are planning on updating them to a later version that fixes these issues before 4.1.

andrejpodzimek, 2019-04-30:
@mikeller I see. It looks like like all my F3s and F4s with BLHeli32 work flawlessly and connect in an instant, but some F7s are affected. Not all of them though:
* `KAKUTEF7` (hexacopter) seems unaffected. No weird configurator delays and BLHeli32 connects just fine.
* `ANYFCF7` (quadcopter) with 20x20mm [Realacc Super_S](https://www.rcgroups.com/forums/showthread.php?2894262-F7-For-your-micro-Now-you-can-Realacc-20X20mm-Super_S-F7-FC) is affected.
* `ANYFCF7` (octocopter) with the 30.5x30.5mm ["classic" board](https://www.banggood.com/STM32F745-100lqfp-216MHz-MPU6000-SPI-F7-Flight-Controller-for-FPV-Racing-Support-Betaflight-p-1137386.html) is affected.

4712, 2019-05-01:
@andrejpodzimek did you try to enable the log function and see where it fails to connect? Actually the in https://github.com/betaflight/betaflight/issues/7873 described bug, mostly is triggered at serial traffic with exactly 64 bytes or more than 64 bytes.

andrejpodzimek, 2019-05-02:
@4712 Just ran your `Test2...` [from here](https://github.com/bitdump/BLHeli/issues/366#issuecomment-487576564) and it generated some logs; here's [one at "level 1"](https://github.com/betaflight/betaflight/files/3136881/log1.log) and [one at "level 4"](https://github.com/betaflight/betaflight/files/3136883/log4.log). (The files should be called `*.xlg` instead of `*.log`, but GitHub doesn't like that.) This is on the `ANYFCF7` quadcopter.

4712, 2019-05-02:
@andrejpodzimek Thanks! Yes, definitely looks like the bug in https://github.com/betaflight/betaflight/issues/7873. After MSP_BOARD_INFO times out the next MSP command gets the prior result at the wrong place. 
Did you try to test connection with Betaflight Configurator?

andrejpodzimek, 2019-05-02:
@4712 Yes, the configurator always connects. But the affected boards exhibit a delay of 2 seconds or the like between the time I click connect and the time a configurator tab shows up. Unaffected boards don’t have this delay.

closed 2019-06-08 by stale[bot]