betaflight/betaflight #8132: Betaflight F7 Flight Controller

cephi58, 2019-04-30:
Why is there no image file for Betaflight F7 Flight Controllers from FPV Model? Is another one fully compatible?

I cannot believe what sort of politics is going around here. Are we customers just jerks you want money from? Another 200 bucks wasted?

RipperDrone, 2019-04-30:
Stop ranting, start reading instead! It took me 2 minutes to find out the matching target .hex file. It's there in the BF4.0 download list.

The board makers have indeed NOT provided a .doc and picture for the Wiki. Pls feel free to add yourself, this is an open source project!

cephi58, 2019-04-30:
thanks. if you know what to look for it is easy, CLI 'version' states BBF7 and there is no such file.
and thanks for your non help. it would have cost you 2 seconds to provide me the name of the correct one instead of the 2 minutes for the one above

RipperDrone, 2019-04-30:
Yeah, sorry for that. Bad attitude seems to create echoes every now and then...

cephi58, 2019-04-30:
pathetic. at least my impression of you guys proved to be right in the end ...

and yes if get your stuff for free you can act that unprofessional way. if you wasted 3000 bucks within one year on non working or doa crap you would see it otherwise.

and handling criticism is obviously not yours ...

mikeller, 2019-04-30:
@cephi58: Why don't you go and complain to the suppliers you spent the money with? They are the ones who should provide you with support, since they took your money.

We are all volunteers donating our time trying to enable others to have a good time, and we did not see any of your money, so your ranting here is way out of line.

cephi58, 2019-04-30:
yeah, you are right if we isolate it to that, and it was me who started to be unfriendly ...

however i encounter often problems updating firmware and a few of your later releases caused some problems resulting in financial damage as well

and i cannot guess which of the board providers do deliver what to you or do not. and in this specific case there was no chance for me to pick up the right hex for reasons already mentioned. and taking the wrong one leads to another 70 bucks damage. getFPV provided me with the info in the meantime that your buddy did not want to give....

and there are defintively some politics between you guys which caused financial damage to us. e. g. the whole Butterflight/Helio comedy costed me >1000 bucks....

that it was exactly a product named Betaflight which is lacking info is maybe not coincidence as well. and, if selfish politics is causing damage to third parties not involved in such sort of quarrels this is not ok to my opinion.

and now let's end the war.

Kind regards
Tommy

mikeller, 2019-04-30:
@cephi58: If you are concerned that using Betaflight is 'causing you damage', you should have read the license you accepted by starting to use Betaflight:

> 16. Limitation of Liability.

>  IN NO EVENT UNLESS REQUIRED BY APPLICABLE LAW OR AGREED TO IN WRITING
WILL ANY COPYRIGHT HOLDER, OR ANY OTHER PARTY WHO MODIFIES AND/OR CONVEYS
THE PROGRAM AS PERMITTED ABOVE, BE LIABLE TO YOU FOR DAMAGES, INCLUDING ANY
GENERAL, SPECIAL, INCIDENTAL OR CONSEQUENTIAL DAMAGES ARISING OUT OF THE
USE OR INABILITY TO USE THE PROGRAM (INCLUDING BUT NOT LIMITED TO LOSS OF
DATA OR DATA BEING RENDERED INACCURATE OR LOSSES SUSTAINED BY YOU OR THIRD
PARTIES OR A FAILURE OF THE PROGRAM TO OPERATE WITH ANY OTHER PROGRAMS),
EVEN IF SUCH HOLDER OR OTHER PARTY HAS BEEN ADVISED OF THE POSSIBILITY OF
SUCH DAMAGES.

If you do not like what you get with Betaflight, please stop using it, and stay away from here. People with your attitude of entitlement are not welcome in open source:

https://mikemcquaid.com/2018/03/19/open-source-maintainers-owe-you-nothing/

closed 2019-04-30 by mikeller

cephi58, 2019-04-30:
yeah, that's what i will definitively do. as 5 out of 7 here did when they abandoned the hobby for exactly those reasons. obviously you guys are completely uncapable of accepting and handling criticism... and obviously you cannot see and accept the whole picture of the fpv business as a complete branch... so what

and i never said anybody of you owes me anything

Closed #8132<https://github.com/betaflight/betaflight/issues/8132>.

mikeller, 2019-04-30:
@cephi58: Good. Then I am sure you don't mind me blocking you from accessing Betaflight again.

Foobin, 2020-01-25:
As a note to future readers who are perhaps new to betaflight and the opensource community in general: The developers working on this project do so out of their own time and commitment to the hobby. They are volunteers, who aren't paid a single cent to maintain, improve and support the firmware that runs on our hardware. Some individuals choose to donate over Paypal or Patreon, but the money that you spent on any hardware running betaflight went directly and solely to the manufacturers of that hardware. 

In light of this, I would say the following concerning @cephi58 remarks, especially considering the fact that they are potentially damaging to the image of the betaflight community:
1. The claim that users of betaflight are 'customers', when by definition a customer implies someone who __buys__ goods/services from a __business__. Neither party can claim to be these things. Users of betaflight do not _buy_ betaflight, and betaflight is not a _business_. This is important to understand, because being a consumer entitles you to certain rights that betaflight is in no position to meet.
2. The claim that the list of supported hardware is being politically influenced by the maintainers: It is true that the betaflight project does not have dedicated, update-to-date firmware for every single flight controller released on the market, however, doing so for any particular FC requires someone to buy it out of their own pocket and then write and maintain the firmware for it for years to come. It is, at best, _unfair_ to ask such a task of someone. Betaflight is open source: like anyone else, if @cephi58 wants firmware for his flight controller, _he can write and maintain it himself_.
3. The claim that updates to the betaflight firmware caused _'financial damage'_: the maintainers of the project do not have access to the exact hardware (incl. motors, escs, radios, sensors) configuration of every particular drone running betaflight. While they do their best to ensure that each major release is as stable as possible, one must accept that bugs can persist. While it is worrying to hear that you are flying at your own risk, remember that there are thousands of people out there doing the same. Problems are typically found and reported early on, listed on the 'Issues' section of this repository, where you can see if someone else with a similar configuration to you has had issues.
4. The claim that more than a thousand (dollars?) were _"lost"_ due to a disagreement over the implementation of a particular algorithm does not stand to reason. For the unaware, "Butterflight" is a fork of the betaflight project, which uses the original implementation of the algorithm in question. I personally cannot discern a single possible cause that could tie this dispute with @cephi58's claimed financial loss.

There are other points which you may have picked up on from @cephi58's posts, but I have skipped over many of them where I find it difficult to discern exactly what he is trying to say. I'll finish up by making the following point: It serves the community __nothing__ to dishearten the developers of the firmware we are using. If it stops being fun, they will simply move on to other projects where their talents and efforts are better recognized and valued, while we are left with an unmaintained codebase. It is, of course, immeasurably important to give feedback to the maintainers so that they can improve upon the project. But make no mistake: there is a very clear line between valuable, constructive criticism and messy, unwarranted conniptions over the internet.

atomiclama, 2020-01-27:
@Foobin unfortunately your well written post will be lost here.