betaflight/betaflight #8139: Decreasing motor PWM Frequency

Difo96, 2019-05-01:
Hello!
My name is Matteo and i'm working on a variable pich quadcopter.
My question is: can i reduce the PWM frequency to control 4 servos instead of 4 ESCs (from 400Hz to 48/50Hz i think)!?
I also have another question but it is marginal, it is not as important as the question above: can i "invert" the PWM signal to invert 2 of the 4 servos!? For example; if the signal to control the ESCs goes from "0" to "100" i need 2 inverted signals that ranging from "100" to "0", to reverse the servos, but if this is not possible i have already bought 5 fisical "singnal inverters"
[image]
[image]

This is my configuration:
FC: Matek F405-CTR
Servos: Tower Pro MG90D 

For all the rest i already found a solution but for these 2 problems i have no idea how to solve them.
Another thing, i am a total programming noob, the only thing that i know to do is open betaflight XD

mikeller, 2019-05-01:
Hi @Difo96. I don't have answers off the top of my head, but there are other users working on variable pitch builds (yours looks very neat by the way). Can I recommend that you join the Betaflight Slack (registration at https://slack.betaflight.com/), and look for user 'Wolfgang S'? He is working on experimental variable pitch aircraft, and he can probably help you.

Difo96, 2019-05-02:
Thanks @mikeller, i'll immediately go to take a look!

closed 2019-05-06 by Difo96

jflyper, 2019-05-06:
Got any insight from Wolfgang S?

Difo96, 2019-05-06:
Yes, i solved; you have to go on Betaflight configurator -> configuration tab, change the ESCs protocol to PWM and separate the PWM frequency from the PID loop. Now you can decreas the PWM freq. The minimum value is 250Hz but most digital servos can support up to 330Hz.

jflyper, 2019-05-06:
👍