betaflight/betaflight #8141: BF 4.01 breaks Colibri Race in QuadQuestions QQ190 RTF

glenb2, 2019-05-02:
Upon flashing from 4.0 to 4.01 the Colibri Race in the QQ190RTF no longer boots.  DFU mode is able to restore to 4.0, which works fine.  I am able to reproduce readily.  

This only occurs on the Colibri Race in this quad.  My other Colibri Race updated fine. Odd.

mikeller, 2019-05-02:
@glenb2: Have you tried flashing with 'Full Chip Erase' enabled?

glenb2, 2019-05-02:
@mikeller I just tried the Full Chip Erase, but got the same result.  No boot and not recognized

mikeller, 2019-05-02:
@glenb2: Thanks for testing. This is weird. Are the two boards the same, and what differences in peripherals connected to the two boards, and in the configurations are there? Can you please post a `diff` for the two boards?

glenb2, 2019-05-02:
# diff
# Betaflight / COLIBRI_RACE (CLBR) 4.0.1 May  1 2019 / 21:36:55 (25f4a5ffe) MSP API: 1.41 / FEATURE CUT LEVEL 1
batch start

feature -LED_STRIP
map TAER1234
aux 0 0 0 1650 2000 0 0
aux 1 1 1 1050 1375 0 0
set gyro_lowpass2_type = BIQUAD
set gyro_lowpass2_hz = 0
set acc_calibration = -23,-74,-21
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = SPEK2048
set blackbox_device = NONE
set motor_pwm_protocol = MULTISHOT
set yaw_motors_reversed = ON
set pid_process_denom = 2
profile 0

set dterm_lowpass2_type = BIQUAD
set dterm_lowpass2_hz = 150
set iterm_rotation = ON
set level_limit = 70
rateprofile 0

batch end

glenb2, 2019-05-02:
diff
Betaflight / COLIBRI_RACE (CLBR) 4.0.0 Apr 11 2019 / 07:03:08 (9ad2cc844) MSP API: 1.41 / FEATURE CUT LEVEL 1
batch start

profile 0

rateprofile 0

batch end

glenb2, 2019-05-03:
I will test this fix for Colibri RACE if a build is posted.

glenb2, 2019-05-05:
Fixed with 4.0.2

closed 2019-05-05 by glenb2