betaflight/betaflight #8142: SPRACINGF3: LEDs broken in 4.0.1 by #8093 [bisected]

andrejpodzimek, 2019-05-02:
TL;DR: Feel free to ignore or close this; it's just for the record for those who want to update their F3s as long as possible. The secret recipe to get a working 4.0.1 with `LED_STRIP` on `SPRACINGF3` is `git revert 774ee5c5a84c10733acf3af95f2ea08739076f65`.

774ee5c5a84c10733acf3af95f2ea08739076f65 breaks `LED_STRIP` on `SPRACINGF3`. This might explain [the question](https://github.com/betaflight/betaflight/issues/8071#issuecomment-485630696) asked by @jflyper in #8071.

I know that F3 support is being phased out, but I have quite many of those and while `LED_STRIP` works just fine with (modified) 4.0.0, 4.0.1 crashes / freezes when `LED_STRIP` is enabled on `SPRACINGF3`. To reproduce, just flash it with full chip erase and then do:

```
feature LED_STRIP
save
```

It will freeze. Sure, LEDs is not supported, but many of my veterans just run a simple hack (after which you remove mags, baros and rangefinders until it fits):

```diff
diff --git a/src/main/target/SPRACINGF3/target.h b/src/main/target/SPRACINGF3/target.h
index 79623030b..b86cf8b82 100644
--- a/src/main/target/SPRACINGF3/target.h
+++ b/src/main/target/SPRACINGF3/target.h
@@ -196,6 +196,9 @@
 
 #define REMAP_TIM17_DMA
 
+#define USE_LED_STRIP
+#define USE_LED_STRIP_STATUS_MODE
+
 // UART1 TX uses DMA1_Channel4, which is also used by dshot on motor 4
 #if defined(USE_UART1_TX_DMA) && defined(USE_DSHOT)
 #undef USE_UART1_TX_DMA
```

(Oh, and if you have an `FF_KOMBINI` and want your motors to work (like, at least one :stuck_out_tongue_winking_eye:), revert that same change; it will help.)

mikeller, 2019-05-02:
@andrejpodzimek: Can you replicate this with officially released firmware? Also, can you please post the information that you were asked to supply when opening an issue, including a `diff` (preferably from official firmware that reproduces the problem, to enable others to replicate)?

andrejpodzimek, 2019-05-02:
I can’t replicate it with officially released firmware, presumably, because that doesn’t enable LEDs. There is officially no `feature LED_STRIP`.

I did provide a diff that replicates the problem `LED_STRIP`is added:

```
feature `LED_STRIP`
save
```

I did not have to add any LEDs or configure anything else to make it freeze. Switching the feature on was all I did.

andrejpodzimek, 2019-05-02:
[Here's a complete `diff all` from one of my F3s.](https://github.com/betaflight/betaflight/files/3138813/diff_all_SPRACINGF3_BRF3_with_LEDs.txt) However, only `feature LED_STRIP` is relevant. With 774ee5c5a84c10733acf3af95f2ea08739076f65, `feature LED_STRIP` freezes the board when switched on and `save`d. Without that change, it flies just fine on this diff (and shines).

jflyper, 2019-05-02:
@andrejpodzimek Your issue may be resolved by the commit in #8148.

0crap, 2019-05-02:
I have the same issue on my OMNIBUS target.
Flashed 4.0.1 on this F3 FC and all is fine.
Configurator opens no problem.
But if I enable LED strip and save, Windows does not recognise the FC anymore.
All I can do is use the BOOT button and flash again in DFU mode for it to start working again.
[image]
[image]

jflyper, 2019-05-02:
@0crap Is this on the official 4.0.1 release firmware?

0crap, 2019-05-02:
Yes it is.
Have reported the same issue in the past. See #7678

jflyper, 2019-05-02:
No, #7678 is completely different.  I'm coming up with a firmware you can test. Wait a sec.

0crap, 2019-05-02:
For me as a user the behaviour is EXACTLY the same as in issue #7678

jflyper, 2019-05-02:
That's why I'm telling you it's completely different. Do you want a fix or not? ;)

jflyper, 2019-05-02:
[betaflight_4.0.1_OMNIBUS-fix timerGetByTag_0.hex.zip](https://github.com/betaflight/betaflight/files/3139770/betaflight_4.0.1_OMNIBUS-fix.timerGetByTag_0.hex.zip)

0crap, 2019-05-02:
Same behaviour different problem, fair enough!
Downloading now, Wait a sec.

0crap, 2019-05-02:
Fix confirmed!
Thank you!

closed 2019-06-08 by stale[bot]