betaflight/betaflight #8143: BF 4.01/Colibri Race (PowerCube) ESCs no longer initialize

glenb2, 2019-05-02:
The first three tones emit, but the last two do not.  4.0 was fine

mikeller, 2019-05-02:
@glenb2: Now I am confused - how does this relate to the problems you opened #8141 for?

IllusionFpv, 2019-05-02:
@mikeller I think that this issue is related to PR https://github.com/betaflight/betaflight/pull/8093
My crazybeef3fr doesn't work too

glenb2, 2019-05-02:
https://www.team-blacksheep.com/products/prod:tbs_powercube_v2

The above linked TBS PowerCube will boot and is recognized, but does initialize the ESCs properly.  Only three tones play on 4.0.1

https://www.team-blacksheep.com/products/prod:tbs_pcv2_2in1
The above linked TBS PowerCube 2in1 with FPVision in the QQ190 will not boot at all with 4.0.1.  Solid red light.  Button initiated DFU mode allows flashing back to 4.0

mikeller, 2019-05-02:
Duplicate of #8141.

closed 2019-05-02 by mikeller