betaflight/betaflight #8149: Paralyze does not disable stick commands

add1ct3dd, 2019-05-02:
**Is your feature request related to a problem? Please describe.**
Kind of - this is for team racing.

**Describe the solution you'd like**
Setup a quad that has Paralyze on an Aux switch.

* turn on quad & radio
* flick paralyze switch so quad is disabled from arming
* flick paralyze switch back to default (or leave it, doesnt matteR)
* either re-arm the quad, or not (again doesn't matter)
* use appropriate stick commands to either access BF OSD, or change profile etc

Outcome: Observe that you can still access the osd etc.

**Describe alternatives you've considered**
There aren't any.

**Additional context**

There is a strong chance to alter some pretty hefty settings in a team race scenario where this feature was developed - in my eyes Paralyze should disable all RX commands, whether this is arming, accessing OSD or any other feature. Currently, this leaves your 'paralyzed' quad open to some very significant settings to be changed while your second quad is flying.

mikeller, 2019-05-02:
Good point, @add1ct3dd, CMS entry should definitely be blocked once paralyse mode is on.

mikeller, 2019-05-04:
Can you please test #8161, @add1ct3dd?

add1ct3dd, 2019-05-04:
Should be able to tonight just at a race :) thank you!

mikeller, 2019-05-07:
@add1ct3dd: How did your tests go? If they were successful #8161 is ready to be merged.

add1ct3dd, 2019-05-07:
I haven't had chance to test yet sorry some things came up & then I forgot - busy weekend!

Will try and test tonight :)

PR #8161 (2019-05-09): Extend the scope of BOXPARALYZE to cover stick commands, adjustments, and CMS.
Fixes #8149.
closed 2019-05-09 by mikeller