betaflight/betaflight #8150: furyf4osd BF >= 4 no gps satcount and homedirection on osd

aknack, 2019-05-02:
everything else is fine. flies like a dream with 401, just those 2 values dont show up in osd.

diff all:

`# diff all

# version
# Betaflight / FURYF4OSD (FY4O) 4.0.1 May  1 2019 / 21:51:35 (25f4a5ffe) MSP API: 1.41

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name FURYF4OSD
manufacturer_id 
mcu_id 003b0046354b500320343246
signature 

# name
name paula der frosch

# resources
resource PPM 1 NONE
resource SERIAL_TX 11 C09
resource ESCSERIAL 1 NONE

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature GPS
feature TELEMETRY
feature LED_STRIP

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 2 2 115200 57600 0 115200
serial 5 2048 115200 57600 0 115200

# led
led 0 7,15::CO:1
led 1 7,14::CO:1
led 2 7,13::CO:1
led 3 8,13::CO:1
led 4 8,14::CO:1
led 5 8,15::CO:1

# color

# mode_color

# aux
aux 0 0 1 1700 2100 0 0
aux 1 1 0 900 1300 0 0
aux 2 2 0 1300 1700 0 0
aux 3 46 3 1700 2100 0 0
aux 4 13 2 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_lowpass2_hz = 100
set dyn_lpf_gyro_max_hz = 570
set acc_calibration = -51,-40,157
set min_check = 1015
set max_check = 1995
set rssi_channel = 16
set rssi_scale = 200
set serialrx_provider = FPORT
set serialrx_halfduplex = ON
set blackbox_device = NONE
set min_throttle = 1015
set motor_pwm_protocol = DSHOT600
set failsafe_procedure = GPS-RESCUE
set current_meter = ADC
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_rescue_angle = 45
set gps_rescue_descent_dist = 100
set pid_process_denom = 1
set osd_warn_core_temp = OFF
set osd_vbat_pos = 2497
set osd_rssi_pos = 2504
set osd_remaining_time_estimate_pos = 472
set osd_g_force_pos = 2522
set osd_mah_drawn_pos = 2508
set osd_gps_speed_pos = 2082
set osd_gps_sats_pos = 2106
set osd_home_dir_pos = 2102
set osd_home_dist_pos = 2098
set osd_altitude_pos = 2089
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set osd_stat_flight_dist = ON
set osd_stat_max_fft = ON
set vtx_band = 1
set vtx_power = 0
set vtx_freq = 5865
set vcd_video_system = PAL
set rcdevice_init_dev_attempts = 4

# profile
profile 0

set iterm_rotation = ON
set smart_feedforward = ON
set angle_level_strength = 65
set horizon_level_strength = 65
set level_limit = 65

# profile
profile 1

set dterm_lowpass2_type = BIQUAD
set dterm_lowpass2_hz = 150

# profile
profile 2

set dterm_lowpass2_type = BIQUAD
set dterm_lowpass2_hz = 150

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 155
set pitch_rc_rate = 155
set yaw_rc_rate = 105
set roll_srate = 72
set pitch_srate = 72
set yaw_srate = 81

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# `

etracer65, 2019-05-02:
Those values won't show up until you have a 3D satellite fix. Have you confirmed that your GPS is working and on the GPS tab in the configurator is shows a 3D fix and latitude/longitude?

aknack, 2019-05-02:
jepp, shows fix and 9 satellites,

Jetrell, 2019-05-08:
@aknack  I'm using a FuryF4OSD with a Beitian BN-180 GPS. Both those show on my OSD.
 I have -
` set gps_auto_baud = OFF`
`Ports sensor input - baud 115200`
 `set gps_rescue_min_sats = 6`

What type of GPS unit are you using? If its only has `GPS` and doesn't have `Glonass, Galileo `etc.  
The GPS unit maybe taking advantage of Ground Assistance.  SBAS can be temperamental sometimes. Which may give a weak ground augmentation lock, that comes in below the default of 8 satellites. even though the reading may show higher. 
Try dropping the default sat count to 6, for a ground test.

closed 2019-06-14 by stale[bot]