betaflight/betaflight #8151: Modified Version of Turtle Mode

Ricofpv, 2019-05-02:
**Is your feature request related to a problem? Please describe.**
Yes. Indoor micros have trouble flipping themselves over on smooth surfaces. They can't get a purchase and just push themselves around upside down. The ability to perform a more axial flip over would help.

**Describe the solution you'd like**
A 2nd version of turtle mode that operates exactly as it already does, but also spins the other two motors in the regular direction. If this is possible.

JyeSmith, 2019-05-02:
Great idea!  Many a time Ive just slid across the floor unable to flip :|

etracer65, 2019-05-02:
I don't really think this would work. How is thrusting around the centerline going to help? The opposite side is still pressing into the floor - now even more. Also the motor direction can't be changed that dynamically with DSHOT commands in the current implementation.

Have you tried setting `crashflip_motor_percent` to a non-zero value? It's designed specifically to help with "whoop" style quads. The idea is that is allows the entire quad to lift which allows rotation since the "down" side isn't hitting the floor.

mikeller, 2019-05-02:
@Ricofpv: Have you tried fixing this with by mechanical improvement? There are good 'grip sprays' that can be used to apply a layer of non-slip rubber to things like hockey sticks. Maybe use some of this on the edges of your prop guards?

closed 2019-06-09 by stale[bot]