betaflight/betaflight #8156: 2S Whoop 4.0.0 Community Presets

maxwellpettit, 2019-05-03:
Hello,

I've recently updated my Eachine Trashcan to the official release of 4.0.0.  On defaults, I've noticed that I get quite a bit of oscillations at low to mid throttle.  I've tried using the brushed/brushless whoop community presets, and this seemed to worsen the behavior for me.  Compared to the defaults of 3.5.7, the 4.0.0 settings behaved much worse.

I would really like to see a preset for 2S whoops that handles better than the 4.0.0 defaults.

I've found that personally, what made the biggest difference for me is disabling the d_min features:
set d_min_roll = 0
set d_min_pitch = 0

Maybe a community preset for 2S whoops should disable this?  

Alternatively, I could see raising the d_min to something closer to the default d_pitch/d_roll.

I've listed my current diff for PIDs/Filters below.  I need some more time to tune this, but I would love to see what settings others are running for 2S whoops on 4.0.0.  I started tuning from the defaults since they behaved much better for me personally.  My filters are also a bit more conservative since I was getting some oscillations.

```
# ----------------------------------------------
# Custom Filter Settings
# ----------------------------------------------

set gyro_lowpass_type = PT1
set gyro_lowpass2_type = PT1
set gyro_lowpass2_hz = 100
set dyn_lpf_gyro_max_hz = 750

profile 0
set dterm_lowpass2_type = PT1
set dterm_lowpass2_hz = 100

# ----------------------------------------------
# Custom PID Settings
# ----------------------------------------------

profile 0
set vbat_pid_gain = ON
set d_pitch = 35
set d_roll = 32
set d_min_roll = 0
set d_min_pitch = 0
```

Please note that feature requests are not 'fire and forget'. It is a lot more likely that the feature you would like to have will be implemented if you keep watching your feature request, and provide more details to developers looking into implementing your feature, and help them with testing.

mikeller, 2019-05-04:
@maxwellpettit: #8043 is addressing part of this (the part that is problematic).
Also, we are currently looking into options to add better support for presets into Betaflight configurator.

maxwellpettit, 2019-05-06:
For those who may be having the same issue, here is the full dump of my settings below.  I was able to test these settings indoors at 50% throttle limit and outdoors at full throttle acro/air mode.  This is running much better than stock 4.0.0 or the whoop community preset, with cool motors and minimal oscillations. 

```
# ----------------------------------------------
# Custom Filter Settings
# ----------------------------------------------

set gyro_lowpass_type = PT1
set gyro_lowpass_hz = 150
set gyro_lowpass2_type = PT1
set gyro_lowpass2_hz = 100

set dyn_notch_range = AUTO
set dyn_notch_width_percent = 8
set dyn_notch_q = 120
set dyn_notch_min_hz = 150

set dyn_lpf_gyro_min_hz = 150
set dyn_lpf_gyro_max_hz = 750

profile 0
set dyn_lpf_dterm_min_hz = 150
set dyn_lpf_dterm_max_hz = 250
set dterm_lowpass_type = BIQUAD
set dterm_lowpass_hz = 150
set dterm_lowpass2_type = PT1
set dterm_lowpass2_hz = 100

# ----------------------------------------------
# Custom PID Settings
# ----------------------------------------------

profile 0

set angle_level_strength = 65

set vbat_pid_gain = ON

set p_pitch = 46
set i_pitch = 70
set d_pitch = 35
set f_pitch = 100
set p_roll = 42
set i_roll = 60
set d_roll = 32
set f_roll = 100
set p_yaw = 35
set i_yaw = 100
set d_yaw = 0
set f_yaw = 0

set d_min_roll = 0
set d_min_pitch = 0
set d_min_yaw = 0
```

closed 2019-06-17 by maxwellpettit