betaflight/betaflight #8163: DShot beacon not working: RX_SET / RX_LOST features swapped?

RipperDrone, 2019-05-04:
Bug seems to be on BF side of code rather than BFC, it seems to me: https://github.com/betaflight/betaflight-configurator/issues/1409

Pls let me know if I should close it in BFC and copy content here to avoid double postings...

etracer65, 2019-05-04:
I just tested it and everything looks correct to me. Beacon behaved as configured.

RipperDrone, 2019-05-04:
Hm, did u actually enable RX_SET beacon only and it worked when flipping a switch associated with BEEPER? Doesn't work here...

RipperDrone, 2019-05-04:
Is there anything FC specific about the DShot beacon code, could it be that it only affects the Aikon F4?

etracer65, 2019-05-04:
> Hm, did u actually enable RX_SET beacon only (keep RX_LOST beacon disabled) and beacon worked when flipping a switch associated with BEEPER? Doesn't work here... It does yellow-highlight the BEEPER active switch range, nevertheless no beacon sounds. It only sounds when I enable the RX_LOST beacon (while RC connection stays active, so no actual RX loss happening). Strange...

Yes, I tested all 4 combinations or both off, each on separately and both on. Worked as expected.

> Is there anything FC specific about the DShot beacon code, could it be that it only affects the Aikon F4?

No.

mikeller, 2019-05-05:
@RipperDrone: Have you done a 'Reset to Defaults' / `defaults` / 'Full chip erase'? Could it be a configuration corruption issue?

RipperDrone, 2019-05-05:
Yes, I always do a full erase PLUS CLI defaults as first command. Will anyways try with latest 4.0.1 or Nightly and latest BFC Nightly again and report back...

RipperDrone, 2019-05-05:
Weird, I can easily reproduce it. But there is some news:

- flashed newer BF version: 
# version
# Betaflight / AIKONF4 (AIK4) 4.1.0 May  4 2019 / 06:04:41 (ba95968e0) MSP API: 1.42

- set RX_SET beacon ON, RX_LOST beacon OFF:
# beacon
Disabled:  RX_LOST

- associated BEEPER feature with AUX1 switch, and (as long as USB is still plugged) BFC shows highlighted BEEPER switch range corretly. Now comes the super-strange detail: When triggering the switch during USB plugged in --> beacon works as supposed, BUT:

- power cycling the quad WITHOUT USB plugged in, then flipping the switch --> no beacon

- adding beacon RX_LOST --> beacon sounds when flipping the switch (RX_LOST shouldn't matter here!...)

diff:
'''# version
# Betaflight / AIKONF4 (AIK4) 4.1.0 May  4 2019 / 06:04:41 (ba95968e0) MSP API: 1.42

# start the command batch
batch start

board_name AIKONF4
manufacturer_id 

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper

# beacon
beacon RX_SET

# map
map TAER1234

# serial
serial 0 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200
serial 31 1024 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1300 1700 0 0
aux 1 2 1 1700 2100 0 0
aux 2 27 0 1700 2100 0 0
aux 3 13 0 1700 2100 0 0
aux 4 35 2 900 1300 0 0
aux 5 49 0 1300 1700 0 0

# adjrange
adjrange 0 0 3 900 1025 0 2 0 0
adjrange 1 0 3 1025 1075 6 2 0 0
adjrange 2 0 3 1075 1125 7 2 0 0
adjrange 3 0 3 1125 1175 8 2 0 0
adjrange 4 0 3 1175 1225 9 2 0 0
adjrange 5 0 3 1225 1300 10 2 0 0
adjrange 6 0 3 1300 1375 11 2 0 0
adjrange 7 0 3 1375 1425 4 2 0 0
adjrange 8 0 3 1425 1475 5 2 0 0
adjrange 9 0 3 1475 1525 1 2 0 0

# rxrange

# vtx

# vtxtable

# rxfail

# display_name

# master
set acc_calibration = 0,44,19
set serialrx_provider = FPORT
set motor_pwm_protocol = DSHOT600
set failsafe_off_delay = 25
set failsafe_throttle = 1070
set failsafe_switch_mode = KILL
set failsafe_throttle_low_delay = 50
set failsafe_procedure = AUTO-LAND
set align_board_yaw = 135
set ibata_scale = 170
set ibata_offset = -2600
set beeper_dshot_beacon_tone = 3

# profile
profile 0

# rateprofile
rateprofile 0

# end the command batch
batch end

Docteh, 2019-05-06:
I threw your modes onto my MATEKF411. Problem is in your mode configuration. If I remove the failsafe item it works. Otherwise beeping doesn't work.

[image]

mikeller, 2019-05-06:
@Docteh: Heh, good find. Having FAILSAFE and BEEPER on the same switch and range will cause FAILSAFE to cancel out BEEPER.

Configuration problem! :stuck_out_tongue_closed_eyes:

RipperDrone, 2019-05-06:
@Docteh - what a surgical precision find, thank you so much. Never thought about it, had both switches on  simultaneously for my last 25+ builds. Then got nervous when reading that DShot telemetry could get some problems when beacon sounds inflight (due to RX_LOST) so I wanted to switch RX_LOST off. This killed the RX_SET beacon for the reason you found so I got confused. 

Will leave both beacon features on now since I found there is a lock for beacon sound when motors are spinning, so my concern was void.

I'm still confused about the dev's purpose in relation to Failsafe and Beacon: When Failsafe / Kill is supposed to switch all motor related functionality off (including DShot beacon), why can the RX_LOSS DShot beacon still sound? What was/is the rationale behind? Does it make sense like it is or should it be modified/fixed in any way?

Docteh, 2019-05-06:
Failsafe is configurable, and can either drop, or fall. Failsafe is available as a flight mode as it is important to understand settings, and if you've got a TaranisQ X7S its easier to put failsafe on a switch, than try and turn off the transmitter in flight. (Receiver still connected)

https://github.com/betaflight/betaflight/wiki/Failsafe-(Flight-Controller)

Maybe you mean paralyze for completely switching all motor functionality off?

RipperDrone, 2019-05-06:
Didn't make myself clear enough, sorry: I put the DShot beacon beeper on the failsafe switch since I wanted to be economic about the number of aux channels (max 4) to keep Taranis latency low (8 channels). 

Intention: Whenever I trigger failsafe (configured as KILL) and the quad drops, I would like it to sound beacon signals to find it when it's hidden somewhere in the grass.

2 questions:

1. As you explained, RX_SET beacon is blocked by active failsafe. Why has it been coded this way? I assume because in failsafe LAND mode, the motors are spinning and therefore cannot sound a DShot beacon tone at the same time. However, if failsafe mode is KILL, all motors are stopped. Beacon could then sound no problem. I guess beacon's been coded always blocked by failsafe for simplicity (?)

2. On the other hand, it seems that when I enable RX_LOST beacon (on top of RX_SET beacon), there is a beacon sound upon failsafe KILL active - even when the radio link is still up. Why is that?

Docteh, 2019-05-06:
For the second question the failsafe flight mode is designed to trigger the failsafe code, regardless of RC status. 

For the first question, I'm less sure. It sounds like RX_LOST takes priority over RX_SET, and so the choice to not beacon for RX_LOST takes priority over RX_SET. Since betaflight is open source you could look through the 

Personally I plan on mounting a beeper on my quad at some point, so I'd enable RX_LOST for that. But its still nice to have a quiet dshot beacon. 

https://github.com/betaflight/betaflight/blob/b0e06d2e2641f8a4a7eff13a9925269456b2680c/src/main/io/beeper.c#L406-L416

mikeller, 2019-05-07:
@RipperDrone:

> I'm still confused about the dev's purpose in relation to Failsafe and Beacon: When Failsafe / Kill is supposed to switch all motor related functionality off (including DShot beacon),

Failsafe was never supposed to 'switch all motor related functionality off', it is supposed to do whatever is appropriate (and configured) in the case of losing the connection to the TX. Sounding the beacon is part of this.

> As you explained, RX_SET beacon is blocked by active failsafe. 

The different beeper tones / sequences have priorities. RX_LOST / Failsafe (i.e. something bad is happening) is prioritised higher than RX_SET (i.e. the pilot has flicked a switch). Makes sense?

RipperDrone, 2019-05-07:
Yes, thx 4 explaining. Just kinda funny that enabling RX_LOST beacon makes the DShot beeper sound when flicking the Failsafe/'RX_SET' Beeper switch. I assumed RX_LOST was supposed to sound the beeper if and only if radio link was lost. Seems it is also related to failsafe triggered.

closed 2019-05-07 by RipperDrone