betaflight/betaflight #8179: some symbols in osd are not correctly displayed (bf4.1)

m013, 2019-05-04:
diff all
Betaflight / MATEKF722SE (MF7S) 4.0.1 May  1 2019 / 22:01:08 (25f4a5ffe) MSP API: 1.41

[BTFL_cli_BX5_20190504_131608.txt](https://github.com/betaflight/betaflight/files/3144913/BTFL_cli_BX5_20190504_131608.txt)

[image]

i am not sure when it happened, but i remeber in was fine on 4.0 at least.
tried different fonts

mikeller, 2019-05-05:
@m013: It looks like your `diff` does not match the version that your screenshot was taken with (which looks like it is a 4.1 development build). If you want to use a development build of the firmware, you also need to use a development build of configurator, and in this case you need to upload a new font from configurator that matches the OSD in the firmware.

m013, 2019-05-05:
oh, i see, thanks. it is my first gps equipped quad, so it seems i just didn't notice.
usually i could fix osd related problems with font re-upload, but not this time.

closed 2019-05-05 by m013