betaflight/betaflight #8183: BF401 not CLI  Set smartport_use_extra_sensors = ON

CpeDratDoPC, 2019-05-05:
### Describe the bug
**BF401 or BF400**  not CLI:  Set **smartport_use_extra_sensors = ON**
=> telemetry ESC Not

BF357 and below, telemetry ESC OK

### Setup / Versions
Kakute F7 AIO
Tekko32 4in1 35A , BlHeli 32,4
Frsky X4R 
Horus X12S FW:OpenTx 2.2.3
Configurátor BF 10.5.1(7c3c3ef9d)

mikeller, 2019-05-05:
@CpeDratDoPC: Can you please supply the information you were asked to supply when opening an issue?

CpeDratDoPC, 2019-05-05:
Nelze použít Nastavit smartport_use_extra_sensors = ON v BF401 nebo BF400, takže pravděpodobně ESC telemetrie nemusí fungovat

closed 2019-05-05 by mikeller

CpeDratDoPC, 2019-05-05:
BF400 or BF401 or BF402 is not known in the CLI
# set smartport_use_extra_sensors = ON
###ERROR: INVALID NAME###