betaflight/betaflight #8185: Consider removing / canceling completely motor stop feature

Saxin, 2019-05-05:
**Is your feature request related to a problem? Please describe.**
Yes, today with Xclass and people building large general purpose quads using BF FW, motor stop feature (and the easy option to enable it) poses serious injury threat if mishandled, I'm sparing to give typical examples - they are very clear to us all.

**Describe the solution you'd like**
Enabling "MOTOR STOP" is possible only via. CLI, maybe consider using two commands, one to enable the option to appear and second the actually enable it.

etracer65, 2019-05-05:
There are lots of settings that if misused can cause serious consequences. I'm not sure why MOTOR_STOP should be treated differently. Besides, as the firmware exists now enabling MOTOR_STOP requires 2 steps - first MOTOR_STOP must be enabled (disabled by default) and Airmode as a feature must be disabled (enabled by default). So it's unlikely the user would enable MOTOR_STOP by accident.

Saxin, 2019-05-05:
I would argue that for MR - motor stop is bad idea under all circumstances, playing safe should be first priority. I don't see any other feature that may induce such severe and probable user injury when on the ground.

digitalentity, 2019-05-05:
Just FYI: we have removed MOTOR_STOP on MR in INAV a while ago, received a huge pushback from users and added it back.

Docteh, 2019-05-05:
I like using MOTOR_STOP. Clueless people can still blindly copy/paste without reading.

mikeller, 2019-05-05:
@digitalentity: We have done the same a while back...

I think the way we have it now (CLI only, need to disable AIRMODE) is effective to stop newbie users or users not knowing what they are doing from accidentally enabling it.

closed 2019-06-11 by stale[bot]