betaflight/betaflight #8189: OSD profile switch in a small channel area

antigravity82, 2019-05-05:
Currently switching between the OSD profiles requires a complete channel. Since many users (also J. Bardwell) save channels for a low latency (8CH on D16), all switches are distributed on 1-4 channels. That's why the OSD profile has to be switchable over a small area. 
For example: 1600 = OSD2, 1800 = OSD3, remaining area OSD1.

I would suggest for this the Modes tab with two adjustable ranges (OSD2 and OSD3). 

For better understanding, here is my switch configuration:
AUX1: Arming
AUX2: Flightmode-switch
AUX3: Beeper-switch, Turtlemode-switch, OSD-switch, Failsafe-switch
AUX4: RSSI

mikeller, 2019-05-05:
@antigravity82: I assume you are talking about FrSky TX hardware? This hardware is perfectly capable of transmitting 4 RC channels and 12 switched channels at 9ms intervals - it's just when you are using firmware on FrSky devices that isn't capable of making use of the hardware that you are running into these problems.
I suggest upgrading to TX firmware that can properly use your hardware, like https://github.com/mikeller/opentx/releases.

antigravity82, 2019-05-05:
@mikeller I do not have gauges to measure the latency of different firmware versions. I can only do what is recommended to me by the Betaflight wiki. 
The 8-CH limit is strongly recommended by betaflight when rc_smoothing_type = FILTER, and so do I: "IMPORTANT: SBus or FPort users should set their radio link to D16 mode and no more than 8 channels." 
https://github.com/betaflight/betaflight/wiki/Tuning-Tips-for-Betaflight-3.4 
https://github.com/betaflight/betaflight/wiki/SBus-FPort-and-RC-Smoothing 

As I said, I and thousands of other users do not want to waste a full channel for the OSD switch. Bardwell has put ALL switches on one channel on his Taranis, and i use tree. Please implement this in the Modes tab in the next version. Greetings from Germany

mikeller, 2019-05-05:
@antigravity82: If you insist on using poorly designed software that fails to use the capabilities of the hardware that you own then this cannot be helped. Closing this.

closed 2019-05-05 by mikeller

mikeller, 2019-05-06:
Updated the betaflight wiki to more accurately reflect on the situation re SBus / FPort capabilities: https://github.com/betaflight/betaflight/wiki/SBus-FPort-and-RC-Smoothing-when-using-OpenTX