betaflight/betaflight #8207: SPRacing F3 do not change to PWM RX Mode with BF4.xx

rolo95, 2019-05-06:
**Describe the bug**
A clear and concise description of what the bug is.
i was using BF 3.5.7 with my FC and all was good with my PWM rx ( flysky FS-IA6)
one wire oer channel
**To Reproduce**
Steps to reproduce the behavior:
please use a SPRacing F3 board fron bangood and flash any 4.00 4.01 or 4.02
and when you change RX mode to PWM ( i wire per channel ) hit save and reboot
the setting do not take the PWM it was stuck in PPM mode. that do not happen with BF3.xx
**Expected behavior**
A clear and concise description of what you expected to happen.
make BF4.xx to take PWM rx settings
**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.
SPRacing F3 
FlySky RX ( FS-IA6) wired in PWM
**Additional context**
Add any other context about the problem here.

jflyper, 2019-05-06:
Please provide the flight controller configuration as instructed above.

rolo95, 2019-05-06:
'''
CLI

# diff
# Betaflight / SPRACINGF3 (SRF3) 4.0.0 Apr 11 2019 / 07:41:11 (9ad2cc844) MSP API: 1.41 / FEATURE CUT LEVEL 6
batch start

beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -RX_SET
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL
map EART1234
aux 0 0 2 1800 2100 0 0
aux 1 1 3 900 1200 0 0
aux 2 2 3 1300 1700 0 0
set gyro_sync_denom = 2
set acc_trim_pitch = -4
set mag_hardware = NONE
set baro_hardware = NONE
set mid_rc = 1499
set min_check = 1005
set max_check = 1950
set airmode_start_throttle_percent = 32
set dshot_idle_value = 450
set motor_pwm_protocol = DSHOT600
set gyro_1_sensor_align = DEFAULT
profile 0

set p_pitch = 50
set i_pitch = 0
set d_pitch = 0
set p_roll = 46
set i_roll = 0
set d_roll = 0
set p_yaw = 65
set i_yaw = 45
rateprofile 0

set tpa_rate = 10
set tpa_breakpoint = 1650
batch end

# diff
# Betaflight / SPRACINGF3 (SRF3) 4.0.0 Apr 11 2019 / 07:41:11 (9ad2cc844) MSP API: 1.41 / FEATURE CUT LEVEL 6
batch start

beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -RX_SET
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL
map EART1234
aux 0 0 2 1800 2100 0 0
aux 1 1 3 900 1200 0 0
aux 2 2 3 1300 1700 0 0
set gyro_sync_denom = 2
set acc_trim_pitch = -4
set mag_hardware = NONE
set baro_hardware = NONE
set mid_rc = 1499
set min_check = 1005
set max_check = 1950
set airmode_start_throttle_percent = 32
set dshot_idle_value = 450
set motor_pwm_protocol = DSHOT600
set gyro_1_sensor_align = DEFAULT
profile 0

set p_pitch = 50
set i_pitch = 0
set d_pitch = 0
set p_roll = 46
set i_roll = 0
set d_roll = 0
set p_yaw = 65
set i_yaw = 45
rateprofile 0

set tpa_rate = 10
set tpa_breakpoint = 1650
batch end

'''

rolo95, 2019-05-06:
```CLI

# diff
# Betaflight / SPRACINGF3 (SRF3) 4.0.0 Apr 11 2019 / 07:41:11 (9ad2cc844) MSP API: 1.41 / FEATURE CUT LEVEL 6
batch start

beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -RX_SET
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL
map EART1234
aux 0 0 2 1800 2100 0 0
aux 1 1 3 900 1200 0 0
aux 2 2 3 1300 1700 0 0
set gyro_sync_denom = 2
set acc_trim_pitch = -4
set mag_hardware = NONE
set baro_hardware = NONE
set mid_rc = 1499
set min_check = 1005
set max_check = 1950
set airmode_start_throttle_percent = 32
set dshot_idle_value = 450
set motor_pwm_protocol = DSHOT600
set gyro_1_sensor_align = DEFAULT
profile 0

set p_pitch = 50
set i_pitch = 0
set d_pitch = 0
set p_roll = 46
set i_roll = 0
set d_roll = 0
set p_yaw = 65
set i_yaw = 45
rateprofile 0

set tpa_rate = 10
set tpa_breakpoint = 1650
batch end

# diff
# Betaflight / SPRACINGF3 (SRF3) 4.0.0 Apr 11 2019 / 07:41:11 (9ad2cc844) MSP API: 1.41 / FEATURE CUT LEVEL 6
batch start

beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -RX_SET
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL
map EART1234
aux 0 0 2 1800 2100 0 0
aux 1 1 3 900 1200 0 0
aux 2 2 3 1300 1700 0 0
set gyro_sync_denom = 2
set acc_trim_pitch = -4
set mag_hardware = NONE
set baro_hardware = NONE
set mid_rc = 1499
set min_check = 1005
set max_check = 1950
set airmode_start_throttle_percent = 32
set dshot_idle_value = 450
set motor_pwm_protocol = DSHOT600
set gyro_1_sensor_align = DEFAULT
profile 0

set p_pitch = 50
set i_pitch = 0
set d_pitch = 0
set p_roll = 46
set i_roll = 0
set d_roll = 0
set p_yaw = 65
set i_yaw = 45
rateprofile 0

set tpa_rate = 10
set tpa_breakpoint = 1650
batch end

```

rolo95, 2019-05-06:
if i flash 3.5.7 i can set the TX to PWM, any of the BF4.xx do not save and commit the PWM change, it stucks to PPM

etracer65, 2019-05-06:
PWM support has been dropped from that target in 4.0 because it is out of flash space. You would have to build a custom version if you wanted to re-enable it (by removing other features).

rolo95, 2019-05-06:
no wonder, or just get a PPM RX, as a matter of fact i was using PWM not because i like to deal with many cables but because that was included when i purchased my FlySki i6 transmitter, many thanks for the fast replies and keep the awesome work you guys are doing for the RC community

swilkens, 2019-05-08:
This was also asked here:

https://github.com/betaflight/betaflight/issues/7951

I uploaded a few .hex files for the SPRacing F3 board with PWM enabled again, you might try those.

rolo95, 2019-05-08:
Swilkens, 
thanks for the pointer, hey but the hex file that is in #7951 reads version 4.1.0 , 4.0.2 is the newest official version, am I missing something?

mikeller, 2019-05-08:
@swilkens: If the feature cut level can be lowered for SPRACINGF3 we can do so in a maintenance release. Please open a pull request for this.

rolo95, 2019-05-08:
how

swilkens, 2019-05-09:
@rolo95 they read 4.1.0 because I built the hex file from the latest code available at the time, which is tagged 4.1.0. You prefer a hex file with 4.0.2?

Please let me know, I can create the hex file for you.

mikeller, 2019-05-12:
@swilkens: I had a quick look at SPRACINGF3 in `4.0.x-maintenance`, and lowering the feature cut level by 1 resulted in this:

```
/usr/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 844 bytes
```

:-(

swilkens, 2019-05-12:
@mikeller yes, PWM is fairly early in the feature cut. FEATURE_CUT_LEVEL makes it easier to maintain but harder to customize, as we are finding now.

Given how popular DSHOT is, I'm still debating myself if it's worth to try and include PWM in the F3 builds. On the other hand, I've built a lot of F3 targets to get PWM back in for various people, heh. I'm currently going through the F3 targets (like I did with the bff3) to find nice feature cut levels at the final supported level for the F3 boards.

I think the best solution will be to just further improve the custom version building documentation and just link other users with questions to that.

I'll make something out of this... maybe if we move some features around in the levels.

I'll rebuild the F3s and see what I get, To be continued.

rolo95, 2019-05-12:
@swilkens
that will be awesome , I use dshoot600 with my Spracing F3 , and yes I think pwm esc mode is almost obsolete as allmost any esc right now support dshoot at least the slower modes , but I was referring as pwm rx , I know that ibus , subs and ppm are prefered and less messy to wire but I was one of the not elite diy builders who got an pwm rx board bundled with the Flysky-I6 , so yes , if you guys can just turn on the pwm rx mode in the BF4.xx builds same as BF3.5.x , I will really appreciate it, I saw the guide to install a Linux distro and compile a custom version with the pwm rx feature enabled but truly I don't have any coding chops to the level of doing all the steps, last time I tried I wiped off my win10 installation . :( , please help the mortals who have not born with the gift of coding ;)

rolo95, 2019-05-12:
ouchhhh... me donkey!! lol 
i miss you reply of the hex file you made
and yes please make one
I can't see 4.1 on my bf configurator flash options only 4.0.x version show 
that's why I get puzzled
but if 4.1 is the latest , yeah!! it is welcomed!!

rolo95, 2019-05-13:
Ahh, i can see BF4.1 in Configurator for the SPRacing F3 but i need to switch to development  DL and show as 4.1 #1566  , it is not in the release list only 4.0.x

mikeller, 2019-05-14:
@swilkens:

> FEATURE_CUT_LEVEL makes it easier to maintain but harder to customize, as we are finding now.

I would disagree. Just add a `#define <feature_name>` to `target.h` for your custom build, and this will override the FEATURE_CUT LEVEL. Or even easier, copy/paste the respective `#define <feature_name>` into the `#if defined(STM32F3)` section, outside of the FEATURE_CUT blocks, in `common_pre.h`.

rolo95, 2019-05-24:
So , if anyone  could compile  an SPRacing F3 4.0.2 with PWM RX enabled  it will be greatly appreciated

closed 2019-06-30 by stale[bot]