betaflight/betaflight #8212: BF 4.0.2/Colibri Race unable to configure LED strip

glenb2, 2019-05-07:
Able to set and save a race profile color in cli, but led configuration tab does not change or save LED configuration on QQ190 RTF

# diff
# Betaflight / COLIBRI_RACE (CLBR) 4.0.2 May  5 2019 / 11:42:05 (56bdc8d26) MSP API: 1.41 / FEATURE CUT LEVEL 1
batch start

map TAER1234
aux 0 0 0 1750 2000 0 0
aux 1 1 1 1750 2000 0 0
set gyro_lowpass_hz = 0
set gyro_lowpass2_hz = 0
set dyn_lpf_gyro_min_hz = 0
set dyn_lpf_gyro_max_hz = 510
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = SPEK2048
set blackbox_device = NONE
set motor_pwm_protocol = MULTISHOT
set yaw_motors_reversed = ON
set pid_process_denom = 2
set ledstrip_race_color = BLUE
profile 0

set dyn_lpf_dterm_min_hz = 60
set dterm_lowpass2_hz = 0
set iterm_rotation = ON
set level_limit = 70
rateprofile 0

batch end

mikeller, 2019-05-07:
@glenb2: Please read the release notes when installing new firmware. It clearly states:

> support for status display through LED strips was removed from F3 based flight controllers, in order to be able to reclaim some flash space. In its place, LED Strip Profiles (except for the 'Status' profile which is not supported on F3) can be used to set LED strips to fixed colours. LED strip profiles are also available on F4 / F7, to enable simple configuration of the LED strip in OSD (#7485);

closed 2019-05-07 by mikeller

glenb2, 2019-05-07:
Sorry about that