betaflight/betaflight #8223: Betaflight 10.5.1 and CrazybeeF4FS 4.0.0 broke my Eachine Trashcan

norrath93, 2019-05-07:
Hello,
i have a major issue with betaflight and my Eachine Trashcan:

After i flashed the firmware to 4.0.0
, my RSSI is constant under 30% (sometimes it says 101-96)
 and i get the error code Bad_rx_recovery evertime i try to launch.

I flashed back to 3.5 but it still broken and back to 4.0.3 and back to 3.5.6 and so on.
I Resetet my Jumper and done all i know.

When i try to start it just flipps and spinns as crazy for a second before it shutsdown and the message Bad_RX_Recover.
In the Betaflight configurator i get also the Bad_Rx_Recovery every 10sec.

I made a full chip erase and bound it again to my Jumper T8SG V2 with Flysky AFHDS 2A.

But now the weird part, i done that 2 weeks ago with another Trashcan (My First) and it was also broken and unusable.
So i talked with Banngood and they send my a replacement and now 2 Weeks later with my 2.Trashcan the same again……

I did everything but i still cant get a good Signal and start.

It cant be a bad RX because it flow right out of the box with my Jumper T8SG v2 twice so...…… something software based must be the issue.

In the Reciever Tab in Betaflight everything works normal and all Settings are normal.

It is the spi Flysky AFHDS 2A reciever build in the fc.

[image]

My 2 Trashcan are both broken Please help.

here my Diff

# version
# Betaflight / CRAZYBEEF4FS (C4FS) 4.0.0 Apr  3 2019 / 13:48:04 (22b9f3453) MSP API: 1.41

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name CRAZYBEEF4FS
manufacturer_id 
mcu_id 0062003d3437511930303630
signature 

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map

# serial

# led

# color

# mode_color

# aux
aux 0 0 0 1300 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set motor_pwm_protocol = DSHOT600
set ibata_scale = 1175
set flysky_spi_tx_id = 55570491
set flysky_spi_rf_channels = 85,23,79,111,87,75,115,109,145,69,99,155,9,33,35,151

# profile
profile 0

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0```

dkijkuit, 2019-08-05:
So far for being backwards compatible :(

skywinder, 2019-08-25:
any updates? WHere can I find original firmware and settings?

mikeller, 2019-08-25:
@skywinder: Contact your supplier or Eachine support for original firmware and settings.