betaflight/betaflight #8224: Remove tenths of a meter on OSD Altitude readout

Jetrell, 2019-05-08:
I have been flying with the new OSD symbols.
The altitude readout has always had tenths of a meter. Which is not accurate, whether with a barometric sensor, and less with GPS. Even with a Delusion of Precision less than 1.0.

I was going to bring this up before.... But the reason I mention this now. Is due to the extra length of the Altitude readout. 
 Would it be an idea to _remove the tenths of a meter with its decimal point_. To shorten the readout?

[image]
[image]

SlipstreamFPV, 2019-05-08:
Second to this, Why not have a configurable decimal place on all readouts?

Amp readout is the same, not accurate to 0.00A so why have those extra numbers?

mikeller, 2019-05-09:
@SlipstreamFPV: Configurable? Not so sure about that - a lot of extra configuration for little benefit. In most cases make a decision of how accurate is accurate enough, then stick with it.

closed 2026-09-26 by blckmn