betaflight/betaflight #8233: AIKONF4 Wrong current readings when using DSHOT

klutvott123, 2019-05-08:
**Describe the bug**
Current sensor shows wrong value when using DSHOT. When leaving the esc protocol at the default ONESHOT125 the current sits at around 0.2-0.3A as expected. As soon as DSHOT is chosen the current jumps up to a higher value. Right now it's at 2.3A but i have seen it as high as 4-5A. Change back to ONESHOT125 and everything is fine again. I'm using the scale from the "manual" I got with my aikon ak32 35a 4 in 1 esc which is 210. This seems to be an issue with several FW versions. Tested with 4.0.2, 3.3.3 and a bunch of jenkins builds previously. 

I have two of these FCs. I'm using the first one with a racerstar 35A 4 in 1 and it has the same issue. I initially thought it was just the esc that was bad, tinkered with the scale/offset a bit but could never get it right.

**To Reproduce**
Flash any BF version between 3.3.3(probably an issue before this too) and 4.0.2. Select DSHOT and watch the current value increase. 

**Expected behavior**
Current sensor should not be affected by the esc protocol chosen.

**Flight controller configuration**
```
# version
# Betaflight / AIKONF4 (AIK4) 4.0.2 May  5 2019 / 11:34:11 (56bdc8d26) MSP API: 1.41

# start the command batch
batch start

board_name AIKONF4
manufacturer_id 

# serial
serial 0 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200

# master
set serialrx_provider = FPORT
set serialrx_inverted = ON
set motor_pwm_protocol = DSHOT600
set ibata_scale = 210

# profile
profile 0

# rateprofile
rateprofile 0
```

**Setup / Versions**
Aikon F4 connected to Aikon ak32 35A 4 in 1/Racerstar 35A 4 in 1
Both have R-XSR receivers using FPORT connected the exact same way. 

No exotic connections for anything. ESC and FC connected with the connector supplied with the esc. Receivers connected to telemetry pad of FC.

etracer65, 2019-05-08:
Add a small capacitor between the current sensor pad an ground. Like 0.1uF.

klutvott123, 2019-05-08:
@etracer65 So crosstalk? I just tried a .1uF ceramic cap and now it works as it should. Seems like this is a design oversight by aikon so I'll just close this. Thanks for the help!

closed 2019-05-08 by klutvott123