betaflight/betaflight #8242: betaflight_4.0.2_OMNIBUSF4SD.hex doesn’t work uart1 rx1 for i-bus

mrSerg87, 2019-05-09:
**Describe the bug**
betaflight_4.0.2_OMNIBUSF4SD.hex doesn’t work uart1 rx1 for i-bus
betaflight_3.5.7_OMNIBUSF4SD.hex everything works.

**Expected behavior**
I would like to get Ibus on rx1 with rssi support as it has worked before.

**Flight controller configuration**
# diff all

# version
# Betaflight / OMNIBUSF4SD (OBSD) 4.0.2 May  5 2019 / 12:08:37 (56bdc8d26) MSP API: 1.41

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name OMNIBUSF4SD
manufacturer_id 
mcu_id 001d00273037471939333631
signature 

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL

# beeper

# beacon

# map

# serial
serial 0 64 115200 57600 0 115200

# led

# color

# mode_color
mode_color 5 0 9
mode_color 5 1 11
mode_color 5 2 2
mode_color 5 3 13
mode_color 5 4 10
mode_color 5 5 3

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1300 1700 0 0
aux 2 27 3 1700 2100 0 0
aux 3 13 2 1700 2100 0 0
aux 4 19 4 1550 2100 0 0
aux 5 30 1 1700 2100 0 0

# adjrange
adjrange 0 0 5 900 1300 12 5 0 0
adjrange 1 1 5 1300 1700 12 5 0 0
adjrange 2 2 5 1700 2100 12 5 0 0

# rxrange

# vtx

# rxfail

# display_name

# master
set mag_hardware = NONE
set baro_hardware = NONE
set max_check = 2000
set rssi_channel = 11
set rc_smoothing_type = INTERPOLATION
set fpv_mix_degrees = 25
set serialrx_provider = IBUS
set airmode_start_throttle_percent = 32
set rx_min_usec = 900
set rx_max_usec = 2100
set min_throttle = 1040
set dshot_idle_value = 450
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 2000
set failsafe_switch_mode = STAGE2
set small_angle = 180
set deadband = 5
set yaw_deadband = 5
set pid_process_denom = 1
set gyro_1_sensor_align = DEFAULT

# profile
profile 0

set p_pitch = 62
set i_pitch = 55
set d_pitch = 37
set p_roll = 50
set i_roll = 44
set d_roll = 33
set p_yaw = 74
set i_yaw = 45

# profile
profile 1

set p_pitch = 50
set i_pitch = 50
set d_pitch = 27
set p_roll = 46
set i_roll = 45
set d_roll = 25
set p_yaw = 65
set i_yaw = 45

# profile
profile 2

set p_pitch = 50
set i_pitch = 50
set d_pitch = 27
set p_roll = 46
set i_roll = 45
set d_roll = 25
set p_yaw = 65
set i_yaw = 45

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 120
set pitch_rc_rate = 0
###ERROR: CORRUPTED CONFIG: pitch_rc_rate = 0###
set yaw_rc_rate = 175
set roll_expo = 10
set yaw_expo = 43
set pitch_srate = 64
set yaw_srate = 40
set tpa_rate = 10
set tpa_breakpoint = 1640
set throttle_limit_percent = 0
###ERROR: CORRUPTED CONFIG: throttle_limit_percent = 0###

**Setup / Versions**
omnibusf4  https://www.gearbest.com/flight-controller/pp_690559.html?wid=1433363 
 i-bus ia6b.

mikeller, 2019-05-09:
@mrSerg87:

> ###ERROR: CORRUPTED CONFIG: pitch_rc_rate = 0###

Your configuration is corrupted, probably as the result of a firmware update that did not properly reset the configuration.

You need to reset it to defaults, and then re-create the configuration from scratch.

mrSerg87, 2019-05-11:
I reset all the settings and manually entered the parameters I needed, but the i-bus did not work on the RX1. Please, help!
# version
# Betaflight / OMNIBUSF4SD (OBSD) 4.0.2 May 5 2019 / 12:08:37 (56bdc8d26) MSP API: 1.41

# start the command batch
batch start

board_name OMNIBUSF4SD
manufacturer_id

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature SOFTSERIAL

# beeper

# beacon

# map

# serial
serial 0 64 115200 57600 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set mag_hardware = NONE
set rssi_channel = 11
set fpv_mix_degrees = 25
set serialrx_provider = IBUS
set blackbox_device = NONE
set motor_pwm_protocol = DSHOT1200
set small_angle = 180
set deadband = 5
set yaw_deadband = 5
set pid_process_denom = 1

# profile
profile 0

# rateprofile
rateprofile 0

# end the command batch
batch end

mikeller, 2019-05-11:
I do not have iBus hardware, so I cannot help you any more beyond this, sorry.

watageek, 2019-05-11:
There is a bug affecting some F4 target. The issue is under investigation. 

There is a workaround to have 4.0.x working on the specific issue. But keep in mind it's solely as a debuging purpose. With the link in #8021 
You can confirm it's solving your issue.

Have a read in #8094 for all the info.

mrSerg87, 2019-05-11:
Dear developers, this information did not help me. On BF 3.5.7 ibus works. Perhaps in BF 4.0.x, the ibus driver was cut to save space, or the new code does not understand ibus.
On omnibus f4 v2, I use osd, esc dshot 1200 and ia6b. FS-i6X sees ia6b that the voltage reading on the FS-i6X is displayed when voltage is applied to omnibus f4 v2. It may help to compile the firmware code for a new one without the support of smart audio and other receivers besides ibus.

atomiclama, 2019-05-13:
@mrSerg87  I'm using BF 4.0.0 with an original board and iBus is working fine. I will try BF 4.0.2 when I get a chance.

Have you tried 4.0.0 maybe that will help narrow your problem down a bit.

maxic62, 2019-05-14:
Hello guys, I confirm @mrSerg87 behavior
It's the same on my side, no way to have a working RX (DSMX - serial RX) with version > 3.5.7
I tried everything (erase, reflash, reconfigure all, multiple different config) without success, there is no activity on receiver tab at all.

@atomiclama : I tried with version 4.0.2 and 4.0.0 and it's the same

Conf:
Model: Mantis85
FC: OMNIBUSF4SD (Supers_F4 Flight controller built-in OSD), STM32F405RGT6 MCU, MPU6000 
ESC:  BLHELI_S 16.5 Dshot600
Tool: betaflight configurator 10.5.1
RX: DSMX (spektrum 2048) (connected to UART1 - serial)

thks for your support, have a good day

atomiclama, 2019-05-14:
Just a thought to the devs, has there been any change to the initialisation of the IO pin? Any change to Pull up / down ? Could a change in this area knock out the RX i.e. put the RX into bind mode or bootmode?

To me it's a difference in HW / IO config / startup timing.

mikeller, 2019-05-14:
@atomiclama: Potentially. #6681 is a candidate for potentially having this side effect.

atomiclama, 2019-05-14:
@mikeller Anything I can provide to help? Logic traces at startup? Can't really test any possible fixes as my setup is ok.

maxic62, 2019-05-14:
@mikeller I will try ! thanks for the feedback

mrSerg87, 2019-05-14:
Thank!
resource pinio 1 b8
set pinio_config = 1,1,1,1
This solved the problem!

closed 2019-05-14 by mrSerg87

mikeller, 2019-05-14:
@mrSerg87: Glad to hear that this fixed it for you.

The bad news is that you got stiffed by GearBest, they sold you a fake OMNIBUSF4, the original OMNIBUSF4 does not have the design fault that makes this workaround necessary.