betaflight/betaflight #8244: Betaflight 4.0.2 LUXF4OSD no OSD

naimadfpv, 2019-05-09:
**Describe the bug**
Flashing Betaflight 4.0 target LUXF4OSD. OSD is working on stock settings. when i put my diff all into it doesnt work anymore.

**To Reproduce**
Flash target LUXF4OSD Betaflihgt 4.0.2 paste diff all

**Expected behavior**
a working OSD. 

**Flight controller configuration**
# version
# Betaflight / LUXF4OSD (LUX4) 4.0.2 May  5 2019 / 12:01:49 (56bdc8d26) MSP API: 1.41

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name LUXF4OSD
manufacturer_id 
mcu_id 0042002b3536471037363130
signature 

# name
name Cinewhoop Skip.F

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature -ANTI_GRAVITY
feature RX_SERIAL

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 13 1 1700 2100 0 0
aux 2 26 2 900 2100 0 0
aux 3 35 3 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set gyro_lowpass_type = PT1
set gyro_lowpass2_type = BIQUAD
set gyro_lowpass2_hz = 0
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = SBUS
set blackbox_p_ratio = 128
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set motor_poles = 12
set yaw_motors_reversed = ON
set pid_process_denom = 1
set osd_vbat_pos = 2368
set osd_current_pos = 2423
set osd_mah_drawn_pos = 2393
set osd_display_name_pos = 2409
set osd_warnings_pos = 2345
set osd_avg_cell_voltage_pos = 2400
set osd_stat_rtc_date_time = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_min_batt = OFF
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_fft = ON
set debug_mode = GYRO_SCALED
set scheduler_optimize_rate = ON
set vcd_video_system = NTSC

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_lowpass2_type = BIQUAD
set dterm_lowpass2_hz = 0
set vbat_pid_gain = ON
set iterm_relax = OFF
set iterm_relax_type = GYRO
set p_pitch = 30
set i_pitch = 0
set d_pitch = 0
set f_pitch = 0
set p_roll = 30
set i_roll = 0
set d_roll = 0
set f_roll = 0
set d_min_roll = 0
set d_min_pitch = 0
set d_min_boost_gain = 0
set d_min_advance = 0

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_expo = 40
set pitch_expo = 40
set tpa_rate = 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save

**Setup / Versions**
 - Multistar AIO F4 (Hobbyking)
 - RX: FRSky Xm+, VTX: HGLRC, Cam: RuncamMicro
 - RX to UART 1; VTX: Vbat Input Voltage, Video Signal to OSD pad; Cam: vbat Input voltage, video to OSD pad

**Additional context**
tried setting up my quad without diff but same problem. 
could not yet figure out which value crashes OSD.

Docteh, 2019-05-09:
I'm surprised you're setting the whole thing up at once. Try just parts at a time. Like why is acc hardware set to none? Why not leave the OSD on auto? Bidirectional dshot is another interesting thing.

naimadfpv, 2019-05-09:
Did you read all the comments? 

I did that. But if i paste my diff it does not work.

naimadfpv, 2019-05-09:
I added some comments so everyone should know that I don’t just put my diff or dump in a new setup! 

This quad already flew with osd, RPM Filters and tuned PIDs with Plasmatree on Bf 4.0.1.

Docteh, 2019-05-09:
I'm still not clear on what you're describing. I figure its one of these two scenarios

First Scenario, to reproduce bug
1. Configure quad
2. fly quad (optional)
3. make a diff
4. wipe configuration
5. apply diff
6. OSD doesn't work

Second scenario
1. Flash 4.0.2
2. Configure quad
3. fly
4. fly
5. fly
6. osd stops working.

mikeller, 2019-05-09:
@naimadfpv: This sounds like it is just a case of your OSD hardware being broken. If you can't get the OSD to work by flashing back to an older version with a config that used to work then it's the hardware. Sorry.

naimadfpv, 2019-05-09:
Lets start from zero. 

I flashed betaflight 4.0.2

I set up everything. Looked through my Google’s as well on the bench and OSD was Setup as i did in GUI. 

Outside:
Tuned PIDs with Plasmatree LOS. 
The quad sounds and flew very well line of sight outside. 
I took my Fatshark on - no OSD. 

Back home i flashed betaflight 4.0.2 to see if my osd hardware was broken. 
I just enabled OSD looked through my Google’s and OSD worked on bench.

I put my diff in CLI hit save. 
Waited for reboot. 
Looked through my google and had no OSD.

At this point i opened this issue. 

So my OSD is working when i flash and just enable betaflight but when i put my settings into betaflight manually or with diff i don’t have OSD. 

I don’t know what the issue is.

Hopefully you understand me otherwise we can close this issue.

etracer65, 2019-05-09:
So when it "worked on the bench" was the quad powered only by battery like in the field? Or was it powered either completely by USB or had both USB and battery connected?

watageek, 2019-05-10:
And you did tried as docteh said. One part at time to pin down where is the culprit and also you tried from scratch without pasting the diff in the cli... right?

And etracer is giving some interesting thought too. Maybe to 5v-3.3v is flacky on the board. Usb is giving a solid 5v.

On a side note Don't be harsh on the staff. They are doing a wonderful job ( on spare time) to give us (mere mortal) some incredible firmware through their hard work.

naimadfpv, 2019-05-10:
Sorry, i don’t want to be harsh. 
I’m not a native English spaeker.

On the Bench I had OSD with usb and lipo power. But also like in the field just with lipo (no USB) 

I will today Setup my quad with every part at at time to pin down where the culprit is.

naimadfpv, 2019-05-10:
Today it’s working. 

I don’t know why. Haven’t touched anything from yesterday till today. 

I sit on my bench, put my Fatshark on. There is OSD as is Setup yesterday.

So I tried to reproduce it. 

Flashed betaflight firmware 4.0.2 with full chip Erase, nothing setup.
After that I put my google on. There is OSD warning and average cell voltage. 
But i did full chip erase. 
There should only be the OSD Warning. 

So i Hit 10 times the restore Settings button but average cell voltage stays in the middle of OSD. 

Today I can’t reproduce it. 
I flashed firmware several times and setup every part at Once. Osd is working. 

I also can put my diff into. Osd is working.

The only thing i can see today is that when I hit full chip erase I can see average cell voltage and OSD warning. 
But there should only be OSD warning because I haven’t setup anything.

evnngsky, 2019-05-10:
Please Check PAL/NTSC.
This difference makes OSD blank.

naimadfpv, 2019-05-10:
I thought that too because docteh asked why I setup ntsc. 

But I can choose ntsc, pal or auto. Osd stays today.

naimadfpv, 2019-05-10:
Maybe just the full chip erase went wrong yesterday.... and there is no issue 🙈 

Im sitting till 10min under my google and searching for the value that causes osd to crash.
But today the osd doesn’t crash....
Wired

etracer65, 2019-05-10:
Closing this as it's either a intermittent hardware issue or a configuration/setup problem.

closed 2019-05-10 by etracer65