betaflight/betaflight #8245: usb_hid_cdc doesn't on MatekF411RX target with 4.0 firmware

ljlzonda, 2019-05-10:
I've got a MatekF411RX board (Betafpv 2S Aio board), with built in SPI receiver for Frsky/Futaba. I run SFHSS mode to bind to my Futaba radio. I've tried 4.0.0 and 4.0.2, neither will the usb_hid_cdc work with liftoff. Liftoff can detect the controller but can't read out the data of each channel.

etracer65, 2019-05-10:
The receiver on the MatekF411RX is only designed to work with Frsky FASST protocol.

mikeller, 2019-05-11:
@etracer65: I think SFHSS, which uses the CC2500, should work with the MATEKF411RX. But I never tested USB HID CDC with an SPI RX, it is entirely possible that there is something stopping this from working.

mikeller, 2019-05-11:
@ljlzonda: Please provide the information you were asked to provide when opening the issue.

etracer65, 2019-05-11:
I just tested Betaflight 4.0.2 (current release) using a real Matek F411-ONE (what the MATEKF411RX target was made for) and HID mode works perfectly. Tested with both Frsky-D and Frsky-X binding. Also tested on both MacOS and Windows 10 using the Liftoff game.

ljlzonda, 2019-05-11:
> I just tested Betaflight 4.0.2 (current release) using a real Matek F411-ONE (what the MATEKF411RX target was made for) and HID mode works perfectly. Tested with both Frsky-D and Frsky-X binding. Also tested on both MacOS and Windows 10 using the Liftoff game.

Thanks mate! I run SFHSS so maybe this issue only happens to SFHSS protocol.

ljlzonda, 2019-05-11:
> The receiver on the MatekF411RX is only designed to work with Frsky ACCST protocol.

In BF 4.0, all spi Frsky receivers are capable with SFHSS protocol, that's really a good news for Futaba Radio owners!

mikeller, 2019-05-12:
@ljlzonda: Verified with your configuration (except for using FRSKY_X instead of SFHSS because I do not have Futaba hardware). Joystick is recognised and works well in linux - it looks like the problem is with your operating system / the simulator.

closed 2019-06-18 by stale[bot]