betaflight/betaflight #8252: CRAZYBEEF3FR 4.0.2 crossfire on UART 3 have nothing showing up in the radio tab worked fine on betaflight 3.5.7

Bammo22, 2019-05-11:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

mikeller, 2019-05-11:
No information provided.

closed 2019-05-11 by mikeller

Bammo22, 2019-05-11:
On all versions of betaflight after V3.5.7 can't get crossfire to work on UART3 . All the settings save in ports and config but when you go to receiver tab roll, pitch, yaw, throttle, Aux 1234 have no values in them.