betaflight/betaflight #8255: Full throttle when changing filter to biquad in reciever when using pt1.BF4.02

promekar, 2019-05-12:
Hello i was messing around with reciever filtering to tune my radio better(radiolink at10II) and i changed the filtering on reciever from PT1 to BIQUAD when armed the quad went full throttle.
Pasting DIFF below 
# version
# Betaflight / DALRCF405 (DLF4) 4.0.2 May  5 2019 / 11:44:41 (56bdc8d26) MSP API: 1.41

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name DALRCF405
manufacturer_id 
mcu_id 003400423037470639343235
signature 

# name
name Remix

# resources

# timer
timer C06 0
timer C07 1

# dma
dma pin A10 1
# pin A10: DMA2 Stream 6 Channel 6
dma pin A08 2
# pin A08: DMA2 Stream 3 Channel 6
dma pin C08 1
# pin C08: DMA2 Stream 4 Channel 7
dma pin C07 NONE

# mixer

# servo

# servo mix

# feature
feature OSD

# beeper

# beacon

# map

# serial
serial 0 64 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1825 2050 0 0
aux 1 35 2 1825 2050 0 0

# adjrange

# rxrange
rxrange 0 1071 1938
rxrange 1 1071 1938
rxrange 2 1071 1938
rxrange 3 1071 1938

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set gyro_lowpass_hz = 0
set gyro_lowpass2_type = BIQUAD
set gyro_lowpass2_hz = 0
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 100
set dyn_lpf_gyro_min_hz = 0
set dyn_lpf_gyro_max_hz = 510
set acc_calibration = 47,81,-105
set mag_hardware = NONE
set min_check = 1030
set max_check = 1980
set rssi_channel = 10
set rc_smoothing_input_hz = 60
set rc_smoothing_derivative_hz = 21
set rc_smoothing_input_type = PT1
set rc_smoothing_derivative_type = PT1
set blackbox_p_ratio = 64
set dshot_idle_value = 580
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set motor_poles = 12
set bat_capacity = 1300
set vbat_max_cell_voltage = 420
set vbat_warning_cell_voltage = 335
set vbat_scale = 109
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 5
set pid_process_denom = 1
set osd_cap_alarm = 1300
set osd_vbat_pos = 14593
set osd_rssi_pos = 14561
set osd_link_quality_pos = 14785
set osd_tim_2_pos = 14807
set osd_crosshairs_pos = 14541
set osd_motor_diag_pos = 14787
set osd_warnings_pos = 14793
set osd_stat_max_spd = OFF
set osd_stat_min_batt = OFF
set osd_stat_endbatt = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_curr = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_fft = ON
set debug_mode = GYRO_RAW
set scheduler_optimize_rate = ON
set vcd_video_system = PAL

# profile
profile 0

set dyn_lpf_dterm_min_hz = 60
set dyn_lpf_dterm_max_hz = 190
set dterm_lowpass_type = PT1
set dterm_lowpass2_hz = 180
set vbat_pid_gain = ON
set anti_gravity_gain = 10000
set feedforward_transition = 5
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 11
set throttle_boost = 8
set p_pitch = 48
set i_pitch = 85
set d_pitch = 58
set f_pitch = 90
set p_roll = 44
set i_roll = 80
set d_roll = 55
set f_roll = 95
set p_yaw = 30
set i_yaw = 0
set d_yaw = 20
set f_yaw = 60
set abs_control_gain = 10
set use_integrated_yaw = ON
set d_min_roll = 35
set d_min_pitch = 38
set d_min_boost_gain = 45
set d_min_advance = 0

# profile
profile 1

set dyn_lpf_dterm_min_hz = 60
set dyn_lpf_dterm_max_hz = 175
set dterm_lowpass2_type = BIQUAD
set dterm_lowpass2_hz = 150
set vbat_pid_gain = ON
set anti_gravity_gain = 10000
set feedforward_transition = 5
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 11
set throttle_boost = 8
set d_pitch = 58
set f_pitch = 85
set d_roll = 55
set f_roll = 95
set p_yaw = 30
set i_yaw = 0
set d_yaw = 20
set f_yaw = 60
set abs_control_gain = 10
set use_integrated_yaw = ON
set d_min_roll = 35
set d_min_pitch = 38
set d_min_boost_gain = 45
set d_min_advance = 0

# profile
profile 2

set vbat_pid_gain = ON
set feedforward_transition = 5
set iterm_relax_type = GYRO
set throttle_boost = 8
set p_yaw = 30
set i_yaw = 0
set d_yaw = 20
set f_yaw = 60
set abs_control_gain = 10
set use_integrated_yaw = ON

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set thr_expo = 10
set yaw_rc_rate = 105
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set roll_srate = 82
set pitch_srate = 82
set yaw_srate = 65
set tpa_rate = 80
set tpa_breakpoint = 1800

# rateprofile
rateprofile 1

set thr_expo = 10
set yaw_rc_rate = 105
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set roll_srate = 82
set pitch_srate = 82
set yaw_srate = 65
set tpa_rate = 80
set tpa_breakpoint = 1800

# rateprofile
rateprofile 2

set thr_expo = 10
set yaw_rc_rate = 105
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set roll_srate = 82
set pitch_srate = 82
set yaw_srate = 65
set tpa_rate = 80
set tpa_breakpoint = 1800

# rateprofile
rateprofile 3

set thr_expo = 10
set yaw_rc_rate = 105
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set roll_srate = 82
set pitch_srate = 82
set yaw_srate = 65
set tpa_rate = 80
set tpa_breakpoint = 1800

# rateprofile
rateprofile 4

set thr_expo = 10
set yaw_rc_rate = 105
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set roll_srate = 82
set pitch_srate = 82
set yaw_srate = 65
set tpa_rate = 80
set tpa_breakpoint = 1800

# rateprofile
rateprofile 5

set thr_expo = 10
set yaw_rc_rate = 105
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set roll_srate = 82
set pitch_srate = 82
set yaw_srate = 65
set tpa_rate = 80
set tpa_breakpoint = 1800

# restore original rateprofile selection
rateprofile 0
[bug.zip](https://github.com/betaflight/betaflight/files/3169918/bug.zip)

etracer65, 2019-05-13:
I can't replicate this issue. What Receiver Mode and Provider are you selecting on the Configuration tab? Nothing is showing up in your `diff` above.

Based on your other reported issue regarding channel glitches it's likely a problem with your TX/RX, but can you provide the exact sequence of events to replicate?

promekar, 2019-05-13:
Im using serial reciever in sbus the the r12dsm diversity reciever(12 channels).
I had rpm filter on and the dynamic dterm lowpass filter on.Had some glitches in control so i tried playing around with the recievers filtering...that was set to 21hz on PT1 and the twitch was ALMOST gone..when i changed the filter in reciever mode the filter from pt1 to BIQUAD and armed i had the quad plugged in usb watching the betaflight traces in reciever tab,i armed and quad and it went crazy(no props of course)...i think it was a weird mix of things...i ll try to replicate too...

etracer65, 2019-05-13:
Well first thing is that we never recommend changing settings in the configurator with a battery connected. When settings are changed there are cases where unexpected behaviors can occur without a reboot.

Along those lines I could possibly see an edge case when switching the filter type without a reboot as the filter BIQUAD filters wouldn't have been initialized because that happens at boot. So it would still take an odd alignment of circumstances, but possible I guess. I can probably change it to be sure that the new settings have no effect until a reboot which is the only way to prevent mixed boot/runtime settings.

promekar, 2019-05-14:
Ok thank you...for your help..keep up the good work...it happened to me and i thought i should let you know in case somebody with props on get this too..:)

Jetrell, 2019-05-16:
I don't want to hijack this thread. But I have a fully throttle issue as well. But under different circumstances.

To start with. I am using Development firmware. And I am using a TBS Crossfire radio. With a 3" machine on 4S. I used both 3S and 4S on this quad. And I am not blaming BF. I'm just after some information, when trying to figure this out.
This may take some explaining.

The issue only manifests itself when on 4S. And at full throttle when the 4S pack voltage drops into the 3S voltage range.
What takes place, is the throttle gets stuck at 99%, when the throttle stick is pulled back.
I had this happen at least 5 times and only under hard climb outs.
It freaked me out the first time. And I was left second guessing whether it actually happened or not!

The odd bit. Is that I have looked through the logs and it all looks fine on the BF side of things. The RX command and FC throttle commands drop at the same time. And the current and audio prop noise all drop off together as well.
All I have to go on is the DVR footage. Its making feel like I'm imagining it!  
But I clearly remember pulling the throttle back on this one specific flight, when I also had to flipped the quad over and pointed it back to earth, because it was at over 200m. And on that DVR footage, the throttle was still at 99% and current was over 50amps. So I know I would never do a change of direction at full throttle like that.

I haven't ruled out the Crossfire Nano. But it does seem centered around when the 4S pack voltage drops into the 3S voltage range. Because it doesn't always happen, and never happens on 3S.
I know the logs show nothing... No excess gyro vibrations at that point.
I'll keep looking and testing....  But can any of you guys think of anything in BF that could cause this? Especially when the log data seems fine?

etracer65, 2019-05-16:
> I don't want to hijack this thread.

Well you did and your question is completely unrelated. If you fell there is a bug then open another issue and provide the requested information.

Jetrell, 2019-05-16:
> 
> 

> 
> Well you did and your question is completely unrelated. If you feel there is a bug then open another issue and provide the requested information.

I don't know enough about this, to know whether it was some how related. So rather than open another issue, without cause, I thought i'd mention it here, in case it was related.

If I get anymore information with respect to this. I will open another issue.
Thanks for the reply :+1:

closed 2019-06-22 by stale[bot]