betaflight/betaflight #8257: Radiolink AT10II twitch(filter smoothing)

promekar, 2019-05-12:
**Describe the bug**
Weird bahavior of quad...sometimes when hovering and apllying a little roll or pitch the two motors twitch and stop right away...this did not happen on 3.5.7 only after 4.0.
**To Reproduce**
Steps to reproduce the behavior:

Tried to reproduce the fault but its completely random.After comparison with the bf3.5 firmware
the rc_smoothing info gives 34mhz...the bf4.0 gives 14mhz 

**Flight controller configuration**
# version
# Betaflight / DALRCF405 (DLF4) 4.0.2 May  5 2019 / 11:44:41 (56bdc8d26) MSP API: 1.41

# start the command batch
batch start

board_name DALRCF405
manufacturer_id 

# name
name Remix

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature OSD

# beeper

# beacon

# map

# serial
serial 0 64 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1825 2050 0 0
aux 1 35 2 1825 2050 0 0

# adjrange

# rxrange
rxrange 0 1071 1938
rxrange 1 1071 1938
rxrange 2 1071 1938
rxrange 3 1071 1938

# vtx

# rxfail

# display_name

# master
set gyro_lowpass_type = PT1
set gyro_lowpass2_type = BIQUAD
set gyro_lowpass2_hz = 0
set dyn_notch_min_hz = 100
set dyn_lpf_gyro_max_hz = 510
set acc_calibration = 46,27,-134
set mag_hardware = NONE
set min_check = 1030
set max_check = 1980
set rssi_channel = 10
set rc_interp = MANUAL
set rc_interp_int = 13
set rc_smoothing_input_type = PT1
set rc_smoothing_derivative_type = PT1
set blackbox_p_ratio = 64
set min_throttle = 1000
set dshot_idle_value = 580
set use_unsynced_pwm = ON
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 16000
set bat_capacity = 1300
set vbat_max_cell_voltage = 423
set vbat_min_cell_voltage = 325
set vbat_warning_cell_voltage = 335
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1

# profile
profile 0

set dyn_lpf_dterm_min_hz = 60
set dyn_lpf_dterm_max_hz = 185
set dterm_lowpass_type = PT1
set dterm_lowpass2_type = BIQUAD
set dterm_lowpass2_hz = 0
set vbat_pid_gain = ON
set anti_gravity_gain = 10000
set feedforward_transition = 5
set iterm_relax = RPY
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 11
set throttle_boost = 8
set d_pitch = 58
set d_roll = 55
set abs_control_gain = 10
set d_min_roll = 35
set d_min_pitch = 38
set d_min_boost_gain = 45
set d_min_advance = 0

# rateprofile
rateprofile 0

set thr_expo = 10
set yaw_rc_rate = 105
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 65
set tpa_rate = 80
set tpa_breakpoint = 1800
**Setup / Versions**
 - Dalrcf405
 - Dalrc engine 40A BLHeli_32 Rev32.6.2
 - VTX,FPVCAM,RECIEVER(sbus-12ch) to FC

**Additional context**
This Happens in both of my quads the same "twitch" totally different setups.
The fc sometimes i get 34hz on autodetect sometimes 14hz
*the log is with interpolation but same thing happens with FILTER too.
[image]
[bf4.0_interpolation_13ms_manual.zip](https://github.com/betaflight/betaflight/files/3170053/bf4.0_interpolation_13ms_manual.zip)

SchonRichtig, 2019-05-13:
Hi,

i had twitching problems with my AT9S to, at first i thought it is a tuning/esc error, but than i noticed it an other quads too.
I installed the newest firmware on the at9s the problem was gone, but only for a few days.
After switching to the X-Lite I never had it again, so in my case it was a problem with the at9s (gimbals). 
Here you can see the behavior at 1:03 min:
[](https://www.youtube.com/watch?v=2FLgIvcgUmw)
The quad twitsches to the right, this only happened on little stick inputs.
This was with BS 3.4.
Maybe you have the same problem?!

etracer65, 2019-05-13:
The only way this can happen is due to glitches coming from the TX/RX. Another clue is the changing frame rate detection that you mentioned. You could try capturing a log with `set debug_mode = RC_SMOOTHING_RATE` to see if there are irregularities in the frame rate from the receiver (shown in debug(0)).

promekar, 2019-05-13:
Ok ill record some smoothing logs see if we can better understand what happend...the weird thing is when i revert to 3.5 i dont have this problem for some reason,My gimbals got replaced just in case.

etracer65, 2019-05-13:
The debug mode above (`RC_SMOOTHING_RATE`) will show us details of the communication between the receiver and flight controller. It would additionally be useful to record a log using `set debug_mode = RC_SMOOTHING` and that will show us more details about the data for a single axis. Note that by default the logging records the roll axis. It that's not where the "twitching" is happening then change `rc_smoothing_debug_axis` as needed.

promekar, 2019-05-14:
[smoothing tests.zip](https://github.com/betaflight/betaflight/files/3177493/smoothing.tests.zip)
debug_mode = RC_SMOOTHING
tests with various changes in filter hz.

etracer65, 2019-05-14:
Rather than spending a bunch of time scrolling through the logs, can you provide timestamps of concern?

promekar, 2019-05-14:
[rc_smoothing_rate_tests.zip](https://github.com/betaflight/betaflight/files/3177701/rc_smoothing_rate_tests.zip)
in jitter.bbl at 0:09/1:12:289
in jitter2.bbl at 00:19/ 00:27

etracer65, 2019-05-14:
It just looks to me like a gimbal problem (most likely) or some other problem with your TX/RX. The raw data from the RX shows the channel spikes so that's what the flight controller is getting from the receiver. I don't see any problems with RC Smoothing and it's working based on the data supplied from the receiver. Below is a screenshot of debug(0) which has the raw (unsmoothed) data directly from the receiver.

[image]

Other than the data glitches, it looks like the frames are consistent at 13ms. I'm not sure why you're trying all these different RC Smoothing settings. No settings are going to improve the behavior of the glitches and you need to solve that with your radio. Just set it back to defaults - AUTO cutoffs and BIQUAD for both input and derivative. If you really want to make it smoother after that (no need as far as I'm concerned) then adjust `rc_smoothing_auto_smoothness` in the CLI. Default is 10 and larger numbers will be more smoothed.

promekar, 2019-05-14:
I ll try to get another gimbal to test or swap my old ones see if this goes away(maybe got a faulty potensiometer or something) i raised an issue cause it was weird that i didnt have this before....
thank you for your time.

closed 2019-06-20 by stale[bot]