betaflight/betaflight #8262: OMNINXT7 won't build with toolchain version 8.3.0

andrejpodzimek, 2019-05-12:
I normally have `GCC_REQUIRED_VERSION=8.3.0` set in `make/local.mk` and build Betaflight with whatever toolchain version my distro (ArchLinux) has by default. This has worked up until recently. (It generates numerous new warnings, but works.)

Somewhere between 4.0.1 and master, `OMNINXT7` started to have an issue that only shows up with the `8.3.0` tool chain, but doesn't affect the default `7.3.1` downloaded by `make arm_sdk_install`.

### 8.3.0

```
Linking OMNINXT7 
Memory region         Used Size  Region Size  %age Used
        ITCM_RAM:       16472 B        16 KB    100.54%
      ITCM_FLASH:          0 GB        16 KB      0.00%
ITCM_FLASH_CONFIG:          0 GB        16 KB      0.00%
     ITCM_FLASH1:          0 GB       480 KB      0.00%
      AXIM_FLASH:        8771 B        16 KB     53.53%
AXIM_FLASH_CONFIG:          0 GB        16 KB      0.00%
     AXIM_FLASH1:      394360 B       480 KB     80.23%
        DTCM_RAM:       27856 B        64 KB     42.50%
           SRAM1:       54124 B       176 KB     30.03%
           SRAM2:          0 GB        16 KB      0.00%
       MEMORY_B1:          0 GB         0 GB      -nan%
/usr/lib/gcc/arm-none-eabi/8.3.0/../../../../arm-none-eabi/bin/ld: obj/main/betaflight_OMNINXT7.elf section `.tcm_code' will not fit in region `ITCM_RAM'
/usr/lib/gcc/arm-none-eabi/8.3.0/../../../../arm-none-eabi/bin/ld: region `ITCM_RAM' overflowed by 88 bytes
collect2: error: ld returned 1 exit status
```

### 7.3.1

```
Linking OMNINXT7 
Memory region         Used Size  Region Size  %age Used
        ITCM_RAM:       15744 B        16 KB     96.09%
      ITCM_FLASH:          0 GB        16 KB      0.00%
ITCM_FLASH_CONFIG:          0 GB        16 KB      0.00%
     ITCM_FLASH1:          0 GB       480 KB      0.00%
      AXIM_FLASH:        8776 B        16 KB     53.56%
AXIM_FLASH_CONFIG:          0 GB        16 KB      0.00%
     AXIM_FLASH1:      407696 B       480 KB     82.95%
        DTCM_RAM:       27856 B        64 KB     42.50%
           SRAM1:       54112 B       176 KB     30.02%
           SRAM2:          0 GB        16 KB      0.00%
       MEMORY_B1:          0 GB         0 GB      -nan%
   text    data     bss     dec     hex filename
 410116    6356   75612  492084   78234 ./obj/main/betaflight_OMNINXT7.elf
Creating HEX ./obj/betaflight_4.1.0_OMNINXT7.hex 
```

andrejpodzimek, 2019-05-12:
Bisected this to 644d5713dfaf144956b1cd962097b8bec20ef773. (But it affects toolchain 8.3.0 **only**, 7.3.1 builds work just fine. Also, while bisecting, I noticed that the change in `ITCM_RAM` usage is somewhat gradual and sometimes **not monotonic**, so `git bisect` could have misidentified the culprit.)

@etracer65, can this be related to the new DShot telemetry link quality? It's not a problem now, but may get hard to fix once Betaflight decides to upgrade the toolchain.

etracer65, 2019-05-12:
Well not exactly related to the DSHOT telemetry quality stats. Just related to anything in FAST_RAM and that's part of it - although not a huge amount. 768 bytes or so I think. The feature can be disabled or we can consider reducing the stats window and granularity - both are configured via defines.

andrejpodzimek, 2019-05-12:
I see. Then it's a mystery to me why a toolchain version change has such an impact on RAM. Maybe it changed some struct alignment or the like.

andrejpodzimek, 2019-05-13:
@etracer65 With toolchain `9.1.0`, the `ITCM_RAM` gets even bigger. The first commit that breaks `OMNINXT7` with toolchain `9.1.0` is acf0af779061f9abfc0cc497086805a61c64b9e8.

closed 2019-06-19 by stale[bot]