betaflight/betaflight #8266: Add GPS rescue Min/Hover Throttle to cell count auto profile switching

Jetrell, 2019-05-14:
Many of us are using 4 and 6 cells on the same 5" quads.   Or 3 and 4 cells on the same 3" quads. 
PR #7516 is very useful for automatically changing PID profiles for this purpose.

Many of you that use GPS rescue and have fine tune it, are aware.  The speed at which the quad descends to the ground, can be slowed to make for a more controlled return. If the `Minimum throttle `and `Hover throttle` are tuned to take into account the quads available thrust at those throttle values.
This can be the difference between the quad coming down like a meteorite. Or coming down at a more sedate speed.

But, because a quad will have more thrust available at the same given throttle value with a higher cell count. This can completely mess up the smoothness in which GPS rescue will operate with those throttle values.

So what I am requesting. Is that GPS rescue `Min throttle` and `Hover throttle` values,  be added to the `Battery cell count auto profile selection`. So those GPS rescue throttle values are automatically set for a specific cell count.

I am also aware that `GPS_rescue_throttle P` value can be affected by cell count as well. But at worse... it will only over shoot the climb out altitude by 20m or so, with a higher cell count.

McGiverGim, 2019-05-14:
Maybe for this is better to use the vertical speed and not the throttle, like it was proposed here https://github.com/betaflight/betaflight/pull/8015 but it was closed in the end.

Jetrell, 2019-05-14:
I agree with @IllusionFpv  reasoning for such an addition.
To me.... I find 3" machines hard to see with the wrong back drop, at any distance. So if the quad is descending at 40Kph, it doesn't give much time to visualize it, and then switch out of GPS rescue, before it hits the ground.  
But if it is only descending at 20kph, it gives twice as much time to orientate the quad and get control again line of sight. This is assuming that line of sight landing is needed. because your VTX channel is stomped on etc.

IllusionFpv, 2019-05-14:
I can work again on this if there is some folks interest

McGiverGim, 2019-05-14:
To me is a good addition. Now the user must configure the min, hover and max throttle values. Using vertical velocity I suppose we can have a general value that works for all and the user don't need to configure it. So if you have time, I think you can reopen the original PR to see if we can get to a merge.

Jetrell, 2019-05-14:
> 
> I can work again on this if there is some folks interest

@IllusionFpv  I most certainly am interested. I agree with your reasoning.

> To me is a good addition. Now the user must configure the min, hover and max throttle values. Using vertical velocity I suppose we can have a general value that works for all and the user don't need [...]

@McGiverGim  They are good points

closed 2019-05-16 by Jetrell