betaflight/betaflight #8270: Color configuration OSD items

simtcrom, 2019-05-14:
**Is your feature request related to a problem? Please describe.**
Currently betaflight OSD items are all in white.
Wont it be nice to have some option to display them in some colors.
For eg, getting Horizon Bars displayed in some bright colors, may help some beginner FPV flyer.

**Describe the solution you'd like**
Have an option to change color of the OSD items.

**Describe alternatives you've considered**
None

**Additional context**
None

mikeller, 2019-05-14:
@simtcrom: Can you elaborate a bit on how you would like to see this done, considering that the hardware used for the OSD overlay is only capable of overlaying black and white?

simtcrom, 2019-05-15:
> @simtcrom: Can you elaborate a bit on how you would like to see this done, considering that the hardware used for the OSD overlay is only capable of overlaying black and white?

Someone educated me on the current chipset and its limitations. I will close the request.

closed 2019-05-15 by simtcrom