betaflight/betaflight #8271: OSD MPH Display Suffix Corrupted

SteveCEvans, 2019-05-14:
I'd expect the OSD to display 'M' after the speed. This isn't what I'm seeing halfway up the left side of the screen below. Is anybody else seeing this?

[image]

commit 0c37925a4e0cec21d0a15b087250fc15530ec7b7

SteveCEvans, 2019-05-14:
I think it's the RT of RTH:?

[image]

mikeller, 2019-05-14:
@SteveCEvans: The fonts for 4.1 hava been updated. You'll have to upload the latest font version, available in the development version of configurator.

SteveCEvans, 2019-05-14:
Thanks. I'll close this issue then.

closed 2019-05-14 by SteveCEvans