betaflight/betaflight #8275: Throttle Curve Auto Calibration

yujanshrestha, 2019-05-15:
**Is your feature request related to a problem? Please describe.**
I am frustrated with inconsistent prop feel and annoying humps in the throttle curve that must be calibrated out. 

**Describe the solution you'd like**

1) Get the quadcopter in a hover a safe distance away.
2) Flip the mode switch to "Throttle Auto Calibrate."
3) I move the throttle from hover to 100% in about 3 seconds.
4) Aircraft derives a throttle curve from my stick movement and accelerometer signal from (3).
5) The data collected from (4) should be sufficient to generate a lookup table that maps throttle value to expected acceleration. This should then be enough to automatically smooth out throttle curve bumps. It should essentially turn the throttle stick input command into a predictable acceleration instead of the unpredictable value (ESC throttle percentage?) it is now. This should make the quad "feel" the same from prop to prop.
6) Land the quadcopter and save the throttle curve via beta flight configurator.

**Describe alternatives you've considered**
Manually tweaking the throttle curve or buying props that feel linear.

etracer65, 2019-05-15:
Most people just learn throttle control and don’t bother with curves.

etracer65, 2019-05-15:
Also any calibration scheme that requires the user to go to full throttle when armed with props on is never going to be implemented.

ledvinap, 2019-05-15:
As a side note, I tried to estimate throttle from motor input using ACC. A lot of blackbox analysis, but I got nothing even remotely plausible.

Docteh, 2019-05-15:
Could you use a blackbox recording to get the data, and then you'd just need to implement a look up table. I think this would make an interesting video, but by itself it doesn't sound like a great idea. Needs the convincing prototype.

mikeller, 2019-05-16:
Good idea to use blackbox data to calculate the parameters. It is already planned to move to parametrisable throttle curves, but even before that it should be possible to calculate an approximation for the desired throttle behaviour based on Betaflight rates, and then set them in the firmware.

yujanshrestha, 2019-05-18:
@ledvinap

> As a side note, I tried to estimate throttle from motor input using ACC. A lot of blackbox analysis, but I got nothing even remotely plausible.

This is as I feared. Was the ACC signal too noisy to be useful? Can you give some more details about why this did not work out as you expected?

ledvinap, 2019-05-18:
It was some time ago. IIRC, the throttle was quite speed dependent and there was difference acceleration depending on flight conditions. The result was roughly quadratic behavior as expected, but with lot deviation between measurement at equal throttle.

closed 2019-05-30 by yujanshrestha