betaflight/betaflight #8277: Extreme yaw oscillations

Ion-FPV, 2019-05-15:
I just built this new quad using the Talon F7 and Talon Gigawatt and I have this issue where I am getting a ton of yaw oscillation for no reason. 

[video (1).zip](https://github.com/betaflight/betaflight/files/3184775/video.1.zip)

It worked during a mini test hover, but when I went outside with it to do some real test flights, it oscillated so much that I couldn't fly it. I've literally quadruple-checked the motor direction/mapping and have switched from 4.0.0 to 4.0.2 to 3.5.7 and the still issue hasn't gone away. I've attached the blackbox logs if that helps.

[Blackbox.zip](https://github.com/betaflight/betaflight/files/3184777/Blackbox.zip)

The correct blackbox log is titled 19/22: 01:02 - 01:04 [00:02]. Could this be a hardware issue? Since it happened after a good test hover, I think that it must be something in the Configurator. I've tried almost everything that I can think of, but I still can't fly this quad.

etracer65, 2019-05-16:
Please provide the `diff` of your settings as required. If you haven't already, configure from a clean full-chip erase - don't paste a diff or dump of old settings. If it still persists it's likely a hardware problem (particularly as it behaves the same across multiple versions).

McGiverGim, 2019-05-16:
As they say, provide a diff. 

Seeing your Blackbox, the yaw is crazy but maybe is produced by too much noise in D in the other axis. 

Use the defaults filters from 4.0.2: change the DTerm LPF 2 from 150Hz to 100Hz. Check that.
You can check too the new filtering for 4.1 described here: https://github.com/betaflight/betaflight/pull/8273

Ion-FPV, 2019-05-16:
[MMD_Diff.txt](https://github.com/betaflight/betaflight/files/3187531/MMD_Diff.txt)
There's the diff. 
Every time I changed firmware, I set it up from scratch. I only kept the dumps for things like this.

I'll try the different filters later today.

dcool8, 2019-05-16:
This looks like your FC orientation is not correctly specified in the configurator.

Ion-FPV, 2019-05-16:
The FC orientation is correct. I have the arrow on the top facing forward like it should be.

Ion-FPV, 2019-05-17:
I tried adding a ton of filtering to see if that would change the situation and it did. I got rid of almost all of the oscillations. Is there a problem with lowering the lowpass minimums other than latency?

McGiverGim, 2019-05-17:
If you low them too much the quad can fly very bad, because you filter some movements that are real and no vibrations. But I suppose that you have not reach that low.

RipperDrone, 2019-05-22:
Have you tried changing ESC PWM freq from 24kHz default to 32kHz or 48kHz - or vice versa? Did wonders for me on a BLH32 ESC build... many other users confirmed same behavior.

closed 2019-06-28 by stale[bot]

fmckeogh, 2025-03-26:
Sorry to necro an old thread, but I'm having the same issue on a Tyro79 (MATEKF411 FC, with very very poor quality motors with shot bearings). I

With all settings as default I get incredibly violent yaw oscillations, even with no props mounted.

If I reduce Gyro Lowpass 1 dynamic low cut off to 5Hz, the oscillations stop and it can hover successfully, but handles weirdly and occasionally jumps around in the yaw axis.

If I raise it even to 10Hz the oscillations return.

Any advice?

nerdCopter, 2025-03-26:
This github is for development, not support.
Please try https://discord.betaflight.com/